nisargjhaveri/WirelessAndroidAutoDongle #88: Argo Trekking 2020 w/ Pi Zero 2w Needs 2 usbs

algoz098, 2024-03-22:
Hi, first of all, amazing piece of work/software you got here! Cheers!

Now, I have put this into a rasp pi zero 2w and connected to a Argo Trekking (Fiat) 2020.

And did not work first, After some debug, I noticed for some reason, it keep restarting the PI needing to restart the process.
Then I tried to put two usbs, one for power, and another one for data, and after that work.

My question is, is this expected?
If so, is there a way to just use one USB from the car?

IF not, is this something anyone has experiencied? Is something I can do to prevent this problem?

nisargjhaveri, 2024-03-23:
This could happen if the it was not able get enough power from the OTG port. But generally that should not be the case for RPi Zero 2w. I've been using it with just the OTG cable connected.

Logs from the device can help confirm, but might be difficult to get the logs, you might have to make some modifications and build.

algoz098, 2024-03-24:
The problem with getting said information it seems the device keeps restarting. Maybe the car is restarting it?

Gdenge, 2024-04-04:
Does the Pi give you any error blinking, or does it literally turn completely off and on again?

algoz098, 2024-04-05:
Is hard to say anything more, but I can see the led of the PI turn off, and then on again, and for a couple seconds bluethooth and wifi disapear.

Theres no ERROR for theres no screen in the pi, nor does it keep the logs.

But, if I put two cables, one for power another for data, any strange behaiviour, stops.

gxaris, 2024-08-23:
I would like to confirm that the same issue is happening with RPi 3A connected to a Fiat 500 (2021)
If i use 2 USB cables connected to the 2 USB ports of my car, the system performs as normal. Using a single USB cable, the system seems to be boot looping.
The issue can be replicated even when plugging the OTG port to the USB of a windows machine

mvincenzi96, 2024-08-27:
As I mentioned in #137
Same happens with my Fiat Cronos S-Design 2021
I measured the output current coming from both the usb outputs of my car
It measured 0.8A

Rpi Zero W requires at least 1A to work properly
And Zero 2W requires at least 1.2A

nisargjhaveri, 2024-08-27:
My understanding was that 0.8A should be enough for RPi Zero W and Zero 2 W.

https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/power-supplies.adoc lists the typical bare-board consumption to be much lower, I somehow assumed it should work with wifi and bluetooth as well with fairly small currents.

mvincenzi96, 2024-08-27:
I made a mistake, based on the link you posted.
RPI Zero W requires at least 1.2A
And Zero 2W requires 2A

nisargjhaveri, 2024-08-28:
As I understand, those values are recommended supplies, but the boards generally work on much smaller power as well.

In any case, this looks like power supply issue only. We should probably look at ways to lower power consumption. The fact that it keeps working even when removing the power cable also suggests that boot might be requiring higher power, and then not so much.

Maybe we should try disabling some Cpu cores, or disabling led (don't know how much it affects) or any other things that can reduce consumption?

nisargjhaveri, 2024-09-05:
Can someone with this issue try out the following couple of things? Might help us solve the issue.

1. If you connect just the single cable instead of USB to PWR port, would it boot and connect bluetooth and Wifi correctly? Android Auto won't work, because of the wrong usb port, but just to understand if the power provided is the only problem or something else as well.
2. Restricting CPU frequency or disabling cores to save power. There are several articles online around this. Couple of them are here that I found interesting, there are a bunch more.
    - [Disabling cores using maxcpu=N](https://www.jeffgeerling.com/blog/2021/disabling-cores-reduce-pi-zero-2-ws-power-consumption-half)
    - [Using powersave governor with CPUFreq](https://ms.codes/blogs/computer-hardware/raspberry-pi-limit-cpu-usage)

bejoysat, 2024-10-14:
> Can someone with this issue try out the following couple of things? Might help us solve the issue.
>
> 1. If you connect just the single cable instead of USB to PWR port, would it boot and connect [...]

I used Pi Zero W and have this issue of reboots while connecting to Wifi. Tried to power from PWR port and it connects to Wifi fine. Then I connected second cable to data port of car to USB port of PiZ. It connects to Android Auto and all works fine. So looks like it is definitely a boot up power issue. Maybe the USB port of the car is not able to supply adequate power?

nisargjhaveri, 2024-10-20:
> Tried to power from PWR port and it connects to Wifi fine.

This is what puzzles me. If the same car port connected to PWR port was able to provide enough power, just changing it to USB shouldn't make much difference. In my understanding the power lines of both usb cables are connected. And we do not even start usb until after wifi is connected. If the same car port word when connected to PWR, it's not car that is not supplying adequet power, no? Obviously I'm missing something here! :)

Grtschnk, 2024-10-30:
I wonder if starting the USB connection makes some cars renegotiate power and/or power consumption spikes at this point and over current trips. I scoped the 5V header on the pi and it would cut out for a whole second.
Tried it with an RPi4 and RPiZero. The zero would work with two cables.