nisargjhaveri/WirelessAndroidAutoDongle #89: AA does not fire up in Sync 3

dvelabs, 2024-03-23:
Hi,

I tried version [v0.3.1](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/releases/download/v0.3.1/raspberrypizero2w-sdcard.img.xz) on an **RPI zero 2w**.

Everything goes as expected but AA does not fire up in car display.
1. I connect phone to Bluetooth
2. WAADongle after a while shuts the Bluetooth off and my phone connects to the Wifi
3. I see in my phone (Redmi Note 11 Pro) an AA message stating that it searches for cars to connect

That's as far as it gets.

If i connect phone through USB then AA fires up in **Sync 3** display

[2.log](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14730628/2.log)

v0d0r, 2024-03-23:
I had exactly the same problem and assumed its Sync 3.  
My issue was 
1. I didn't use the OTG port 
2. I changed my cable then it started working

sher1ff, 2024-03-23:
I had the same problem as you with a Pi Zero 2W and same three lines in my log

``` 
Jan  1 00:00:58 buildroot daemon.err bluetoothd[163]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
Jan  1 00:00:58 buildroot daemon.err bluetoothd[163]: src/profile.c:ext_io_disconnected() Unable to get io data for HSP HS: getpeername: Transport endpoint is not connected (107)
Jan  1 00:00:58 buildroot daemon.err bluetoothd[163]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
```

As v0d0r stated changing the cable worked for me. Alot of microUSB cables are not "data cables" and are charge only

nisargjhaveri, 2024-03-26:
@sher1ff These three lines that you mentioned are generally harmless and shows up even on successful connection.

@dvelabs Any luck with a different cable and the correct port? Seems like it should be working in Sync 3.

v0d0r, 2024-03-27:
Though mine is detecting and working mostly every time I have an entirely different problem.  I havent logged an issue yet because I have not had much time to properly test and troubleshoot. But my sound is very choppy when listening to music (spotify).  It starts off fine and then after a minute it randomly starts breaking up almost like there is load on the pi or something. So would be curious to know once you get your sync 3 working if you experience something similar. Currently building/moving house so difficult to find time for anything but I will test in more detail eventually and log a issue if needed.

dvelabs, 2024-03-28:
> I had exactly the same problem and assumed its Sync 3. My issue was
> 
> 1. I didn't use the OTG port
> 2. I changed my cable then it started working

Tried what you suggested and worked as expected.
Wish i had tried it with order mentioned.

After swapping all the microusb cables available (borrowed also a few), it snapped me your comment about the OTG port.

Was certain that the OTG port is the one same for charging...

I plugged in the original cable to the other port and snap it worked.

Thanks a lot for your help

NabeelUppel, 2024-04-04:
Do you have two USB cables connected? One for charging and another for the OTG cable?

nisargjhaveri, 2024-04-04:
> Do you have two USB cables connected? One for charging and another for the OTG cable?

Two cables should generally not be required, only the otg cable should be able to provide enough power in most cases.

nisargjhaveri, 2024-04-04:
Closing since the original issue seems to be resolved.

closed 2024-04-04 by nisargjhaveri