nisargjhaveri/WirelessAndroidAutoDongle #91: Add ability to use with arduino nano 33 IoT

L3Prichaun13, 2024-03-24:
I would like this project to be able to be installed onto an Arduino nano 33 IoT.  What is needed in order to do this?
I am a developer but the microcontroller arena is new to me. If anyone can assist with what is needed, I am ok with doing a lot of the leg work and can post the updates to this repo as well. Ultimately I would like this to become more portable to any microcontroller that meets the minimum requirements

admiralspeedy, 2024-03-24:
Impossible, the CPU on that thing is way too weak to do anything close to this.

L3Prichaun13, 2024-03-25:
Thank you for looking at my issue and replying... If you don't mind, can
you show me how you figured that out? I am not asking as to prove your
right but rather so I can learn from this and learn how to figure this out
on my own next time

admiralspeedy, 2024-03-25:
It uses a 48 MHz SAMD21 as it's main processor, with a separate chip for handling wireless connectivity.

It's meant for small and relatively simply IoT projects handling sensors, like alarm systems and stuff like that, and relaying that information wireless. It is not capable of running Linux (which is what this project runs in on a Pi).

L3Prichaun13, 2024-03-27:
Arduino is not capable of running this type of program...Not possible

closed 2024-03-27 by L3Prichaun13