nisargjhaveri/WirelessAndroidAutoDongle #92: Lock keyboard while driving settign is enabled even when disabled from settgin in head unit

subu28, 2024-03-24:
[image]

I have the exact same headunit just as the one on the screenshot. Probably the same you use for testing. I had disabled the lock and was able to use the keyboard full time.

After using the wireless dongle this no longer works. The keyabord is locked and i have to use only voice while moving. I would like to get it working again and contribute if you can get me started.

admiralspeedy, 2024-03-24:
I don't think that has anything to do with this project at all, that's a head unit specific feature (and I don't even know how that works because Android Auto handles the keyboard, not the head unit).

subu28, 2024-03-24:
The headunit can send a disable all restrictions instruction message to the connected android auto on phone. See `sendDrivingStatusUnrestricted()` in `https://github.com/f1xpl/openauto/blob/development/src/autoapp/Service/SensorService.cpp`

I did additional testing in my car just now. I am able to use the keyboard while moving if I enable locking and then disable again after the phone is connected. In the wired mode I did not have to do this each time.

My thesis is that my head unit sends the specific signal early in the connection process and is getting lost on the way somewhere. I can get to the bottom of this with some reading and understanding of this project. Because I have a workaround now, i can live longer with it.

I need to allow typing while driving because names of places here(india) are not going to be understood by the android assistant. I would like my co passenger to be able to add navigation for me.

I think @nisargjhaveri drives a car with same headunit but I could be wrong. This feature I think is not available in most headunits so I dont think other contributors would be able to help. I will not mind if this issue is closed, I will work on it on my own slowly

subu28, 2024-03-24:
I see this is a very specific thing for me. I will mostly fork and solve it for myself instead of bothering anyone here.

closed 2024-03-24 by subu28

nisargjhaveri, 2024-03-24:
If you do find the issue or fix with this repository, do let us know! :)