nisargjhaveri/WirelessAndroidAutoDongle #95: Jeep Cherokee 2019 - Device Not Supported

L3Prichaun13, 2024-03-28:
I just got the SD image created for my raspberry pi 02w (big thanks to admiralspeedy)... I hooked it up to my Jeep Cherokee and was able to connect to the  AA Wireless Dongle via bluetooth. But then my headunit showed a message saying "Device not supported"
I did also notice that within the bluetooth settings for the AA Wireless Dongle, that it showed as a bluetooth earpiece and only had an option for calling and not audio... 
I am more than happy to take a look at this on my end as well to see what I can find out, if anyone can point me in the right direction of where/what I need to gather about my car make to integrate it

nisargjhaveri, 2024-03-28:
It would help if you can share the logs.

L3Prichaun13, 2024-04-13:
Perfect... I will get those for you.

nisargjhaveri, 2024-08-26:
Closing this issue for now. Feel free to reopen with more info if this is still an issue.

closed 2024-08-26 by nisargjhaveri

yungp00t, 2025-07-28:
I have a 2021 Cherokee and eventually had this issue. first SD card took forever to work but worked for maybe 10 seconds then never again, heck couldnt even get AA to work wired after that. Next day i tried a new SD card and it kept rebooting the pi, so i used two micro USB cords and it woul;d stay on and show this same error.

manio, 2025-07-28:
> heck couldnt even get AA to work wired after that

In this case I think it is some bigger problem completely not related with this project :(

As always - hard to tell anything without the logs...