nisargjhaveri/WirelessAndroidAutoDongle #97: Building image

Inrovero, 2024-04-02:
Hi there!

I'm kind of newbie with raspberry, i've already used arduino but it's kinda different, i've downloaded a pre-built image for my raspberry pi zero  2w, everything's ok but i haven't changed my country code neither my wifi channel, so i get a little laggy using it.

Now i'm trying to build my image following the "Build with manual setup" section of the Readme, so i can change those parameters and try the new build, i'm using Powershell in windows, i've already have the make command, but when i run it, i'm stucked getting this error:

"make: *** No rule to make target 'raspberrypizero2w_defconfig'.  Stop."

I'm shure that i'm in the "buildroot" directory, and i've downloaded the "WirelessAndroidAutoDongle-0.4.0.zip" file of source code.

Line of command: make BR2_EXTERNAL=../aa_wireless_dongle/ O=output/raspberrypizero2w raspberrypizero2w_defconfig

Also i've already created the "output" directory inside of "buildroot".

Hope somebody can help me!
Regards,

admiralspeedy, 2024-04-02:
The country code has no impact on performance.

You cannot build through Powershell, and you need more than just the make command to build.

Inrovero, 2024-04-02:
Omg! Really? I didn't noticed that! You saved me, thank you!
Ps. I'm being sarcastic.

Don't you think that I've know that and that's why i made this post?
Please, help or let others help me, don't be such..

xoxo

nisargjhaveri, 2024-04-03:
I don't think it would be trivial to build from powershell. There is some discussion in #94 which might be relevant.

> I'm shure that i'm in the "buildroot" directory, and i've downloaded the "WirelessAndroidAutoDongle-0.4.0.zip" file of source code.

If you download the .zip, do you also get the buildroot directory? I've never tried that. In any case, if possible you can try to follow the `git clone` command mentioned in the README. You need to do a recursive clone to get contents of the buildroot folder as well.

I'd be very happy if someone can experiment and find a way to build with docker properly on Windows. Powershell might be more trickier or might not be possible. In any case, the current known solution is to clone and build with WSL or other Linux VM on Windows.

Also, if you're only trying to change the country code and/or the wifi channels, you might not need to build locally. You can boot up the Pi and try to ssh into it and change the contents of /etc/hostapd.conf.

While you should really change the wifi country code to make it correct, and there are some cases where it can directly affect some parts of wifi performance, not sure that would be the case here. On the other hand, changing wifi channel can have huge impact on wifi performance based on inteference from other Wifi networks.

closed 2024-08-25 by nisargjhaveri