nisargjhaveri/WirelessAndroidAutoDongle #98: Dongle Mode Initial Setup Loop

admiralspeedy, 2024-04-02:
I switched to 0.4.0 this morning to try the new connection fixes with dongle mode and immediately had issues.

The very first connection took nearly a full minute before Android Auto started on my head unit, and then it would immediately freeze, time out and restart over and over. I had to unplug the dongle and plug it back in and now it appears to be working fine.

I was unable to get logs since I was driving but I suspect it has something to do with the connection process of dongle mode and the newly introduced timeouts because it seemed to be freezing and timing out on the same interval (probably 10 seconds).

Inrovero, 2024-04-03:
Try changing your car or your phone, it works like a charm for me.

Maybe the problem is you, try it out !

th873, 2024-04-03:
I have the same problem after upgrading too :/

admiralspeedy, 2024-04-03:
> Try changing your car or your phone, it works like a charm for me.
> 
> Maybe the problem is you, try it out !

See, I'm here to actually assist with the development of this project. I answered your question on #97 properly and there is absolutely no reason to be salty with sarcastic comments like this.

Nobody here is going to spoon feed you instructions on how to actually set up your development and building environment.

Ioniq3, 2024-04-04:
Same problem, the first connection is normally ok, but after turn off the car and turn on 5 minutes after they enter into loop.
I have some extra features, so some lines of the logs are not the usual regarding img mount.

admiralspeedy, 2024-04-04:
@Ioniq3 you jinxed me. I only had this issue the first time I set up dongle mode and it never did it again, but this morning when I was leaving for work shortly after you commented, I got in my car and it kept looping again until I unplugged it and plugged it back in.

I changed the title to reflect that it isn't just during startup.

NemesisXB, 2024-04-06:
First time user. I have the same issue.

tiliarou, 2024-04-07:
didn't fully test it but after upgrading, I had to associate the bluetooth again (I assume it's the normal behavior), then it started to work properly and loaded quite fast in my Renault (with R link 2).
It seems to work properly.
My use case is that I always try to unplug the usb since it seems my car always powers the usb port even when it's locked and I'm away from the car...

Ioniq3, 2024-04-08:
> didn't fully test it but after upgrading, I had to associate the bluetooth again (I assume it's the normal behavior), then it started to work properly and loaded quite fast in my Renault (with R [...]

Yes, modern cars remain 10 , 15 minutes the power in the accesory line after close the car

tiliarou, 2024-04-08:
> Yes, modern cars remain 10 , 15 minutes the power in the accesory line after close the car

Oh ok, I see.
Today I tried a trip with 0.4 and in the middle android auto disconnected. I reconnected the wifi manually and it worked back right away.
Not sure what's happening, if it's the 0.4 version or if it's something else, it never happened with previous version...

tiliarou, 2024-04-08:
I'm having more issues with some disconnect sometimes. Couldn't get log yet.
When I check the WiFi networks I can see 2 with exactly the same ssid, if I disable WiFi, it re-enables (as it should) automatically after some time, and then it works again properly when connecting to the WiFi network

admiralspeedy, 2024-04-08:
@tiliarou Not sure this relates to the issue here.

I have had no connections issues other than as described in this issue. Three times now I have had to unplug the dongle and then plug it back in because it got stuck in a loop of connecting and disconnecting when I got in my car. I have never had any random disconnects.

tiliarou, 2024-04-08:
> @tiliarou Not sure this relates to the issue here.
>
> I have had no connections issues other than as described in this issue. Three times now I have had to unplug the dongle and then plug it back in [...]

Not sure as well... But it never happened with the previous version... Never a single "android auto has encountered an error" prior to 0.4

Ioniq3, 2024-04-09:
> Same problem, the first connection is normally ok, but after turn off the car and turn on 5 minutes after they enter into loop. I have some extra features, so some lines of the log are not the usual regarding img mount. [messages.zip](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14863640/messages.zip)
> 
> I made this changes to get a second test, revert in [6e6a1da](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/commit/6e6a1dae00dce7dc29be3095eb0717121bbece9b) commit the 110 line :
> 
> ```c++
> - disableGadget(accessoryGadgetName);
> ```
> 
> in aa_wireless_dongle/package/aawg/src/usb.cpp
> 
> and increase in 184 line:
> 
> ```c++
>     struct timeval tv = {
>         .tv_sec = 10,
>         .tv_usec = 0,
>     };
> ```
> 
> to 30 sec to permit more idle time and reduce the loop in aa_wireless_dongle/package/aawg/src/proxyHandler.cpp

I test this changes and no longer apper the loop, but I prefer that Nisarg review

tiliarou, 2024-04-09:
I can't find a way to retrieve logs... How do you get to ssh to the raspberry from android ? Target IP shall be 10.0.0.1, right ? On port 22 ? I get a timeout using juicessh on my phone...

admiralspeedy, 2024-04-09:
> I can't find a way to retrieve logs... How do you get to ssh to the raspberry from android ? Target IP shall be 10.0.0.1, right ? On port 22 ? I get a timeout using juicessh on my phone...

I only tried SSHing once from my phone but was unsuccessful because I think the Android classifies the network differently knowing that it's an AA network. To pull logs I connect to the dongle's network from my PC and SSH in that way, which is a bit tough to do because I either have to have a laptop with me or be in my driveway so I can connect from my desktop in my house.

Ioniq3, 2024-04-09:
> I only tried SSHing once from my phone but was unsuccessful because I think the Android classifies the network differently knowing that it's an AA network. To pull logs I connect to the dongle's [...]

I use daremote from Google play. You have to disconnect the WiFi ssid of Android auto and connect to the wifi, same ssid/name but diferentes connection.

tiliarou, 2024-04-10:
Finally managed to get logs... It's very difficult since android auto automatically connects to the WiFi which can't SSH.
A debug flag with log writing would be so good. Off by default to not always write on the sd card.

```
# nano /var/log/messages -sh: nano: not found # cat /var/log/messages Jan  1 00:00:02 buildroot syslog.info syslogd started: BusyBox v1.36.1 Jan  1 00:00:02 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-03-30 17:22:14 UTC) Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0 Jan  1 00:00:02 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61-v7 (runner@fv-az735-94) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP Sat Mar 30 17:55:00 UTC 2024 Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] CPU: div instructions available: patching division code Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero 2 W Rev 1.0 Jan  1 00:00:02 buildroot kern.notice kernel: [    0.000000] random: crng init done Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writealloc Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Zone ranges: Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x0000000019bfffff] Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]   Normal   empty Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Movable zone start for each node Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Early memory node ranges Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff] Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff] Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] percpu: Embedded 12 pages/cpu s18004 r8192 d22956 u49152 Jan  1 00:00:02 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s18004 r8192 d22956 u49152 alloc=12*4096 Jan  1 00:00:02 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545 Jan  1 00:00:02 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:EF:DA:EE vc_mem.mem_base=0x1ec0000 Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear) Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear) Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Memory: 337224K/421888K available (9216K kernel code, 718K rwdata, 2316K rodata, 1024K init, 520K bss, 19128K reserved, 65536K cma-reserved) Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1 Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation. Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]      Tracing variant of Tasks RCU enabled. Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies. Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16 Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention. Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys). Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns Jan  1 00:00:02 buildroot kern.info kernel: [    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns Jan  1 00:00:02 buildroot kern.info kernel: [    0.000018] Switching to timer-based delay loop, resolution 52ns Jan  1 00:00:02 buildroot kern.info kernel: [    0.000344] Console: colour dummy device 80x30 Jan  1 00:00:02 buildroot kern.info kernel: [    0.000854] printk: console [tty1] enabled Jan  1 00:00:02 buildroot kern.info kernel: [    0.000906] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000) Jan  1 00:00:02 buildroot kern.info kernel: [    0.000947] CPU: Testing write buffer coherency: ok Jan  1 00:00:02 buildroot kern.info kernel: [    0.001005] pid_max: default: 32768 minimum: 301 Jan  1 00:00:02 buildroot kern.info kernel:  [... cut ...] 0: STA 28:c2:1f:f4:73:ea WPA: pairwise key handshake completed (RSN) Jan  1 00:06:31 buildroot daemon.info dnsmasq-dhcp[226]: DHCPREQUEST(wlan0) 10.0.0.6 28:c2:1f:f4:73:ea Jan  1 00:06:31 buildroot daemon.info dnsmasq-dhcp[226]: DHCPACK(wlan0) 10.0.0.6 28:c2:1f:f4:73:ea S21-Ultra-de-Robin Jan  1 00:06:54 buildroot authpriv.info dropbear[280]: Child connection from 10.0.0.6:49884 Jan  1 00:06:55 buildroot authpriv.notice dropbear[280]: Password auth succeeded for 'root' from 10.0.0.6:49884 #

```

Ioniq3, 2024-04-10:
> > Same problem, the first connection is normally ok, but after turn off the car and turn on 5 minutes after they enter into loop. I have some extra features, so some lines of the log are not the usual regarding img mount. [messages.zip](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/14863640/messages.zip)
> > 
> > I made this changes to get a second test, revert in [6e6a1da](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/commit/6e6a1dae00dce7dc29be3095eb0717121bbece9b) commit the 110 line :
> > 
> > ```c++
> > - disableGadget(accessoryGadgetName);
> > ```
> > 
> > in aa_wireless_dongle/package/aawg/src/usb.cpp
> > 
> > and increase in 184 line:
> > 
> > ```c++
> >     struct timeval tv = {
> >         .tv_sec = 10,
> >         .tv_usec = 0,
> >     };
> > ```
> > 
> > to 30 sec to permit more idle time and reduce the loop in aa_wireless_dongle/package/aawg/src/proxyHandler.cpp
> 
> I test that changes and no longer appers the loop, but I prefer that Nisarg review

No, this morning made the loop in the first setup, so the problem have to be other issue.

th873, 2024-04-11:
What are you guys running it on? I am using pi zero w. was thinking if that could have something to do with it?

Inrovero, 2024-04-11:
Hi everyone, I'm using the 4.0 version for 1 week now, and I'm almost having no issue, just sometimes but if i reconnect my bluetooth or wifi, it started working like a charm, but just 2 or 3 of 10 times that i get in the car..
I'm using raspberry pi zero 2w + Samsung Galaxy S20fe

I'm having some laggy experience in audio or touchscreen using spotify + maps, but that's another issue in discussion.

I'll try to downgrade to 2.0 version to test this issues

tiliarou, 2024-04-11:
> Hi everyone, I'm using the 4.0 version for 1 week now, and I'm almost having no issue, just sometimes but if i reconnect my bluetooth or wifi, it started working like a charm, but just 2 or 3 of 10 [...]

My setup is quite similar, Raspberry pi zero 2W + Samsung galaxy S21 ultra.

Same issues for me, sometimes it works like a charm, sometimes I have to disable wifi and let it reconnect, sometimes I have to unplug and plug again the raspberry... The point is to have a reliable dongle, so it's still an issue even if it's workable with these problems.

admiralspeedy, 2024-04-11:
> What are you guys running it on? I am using pi zero w. was thinking if that could have something to do with it?

I am using a Pi 0 W as well. Also, I think my Pi died this morning because it is no longer appears as a BT or Wi-Fi device for me despite lighting up when powered on. Checked the SD card and it appears fine.

Inrovero, 2024-04-11:
> I am using a Pi 0 W as well. Also, I think my Pi died this morning because it is no longer appears as a BT or Wi-Fi device for me despite lighting up when powered on. Checked the SD card and it [...]

Have u tried giving usb power in the power connector and connecting the otg to the car? Maybe the power of the otg car isn't enough now.. idk, just saying!

admiralspeedy, 2024-04-11:
> Have u tried giving usb power in the power connector and connecting the otg to the car? Maybe the power of the otg car isn't enough now.. idk, just saying!

Nah, I've been using it for like 2 months and I can't get it to show up when plugged into a USB port on my work PC either.

I'm going to re-flash the SD card and see if something did get messed up there (it's an old SD card and it gets pretty cold in my car at night still).

admiralspeedy, 2024-04-11:
Yah, it was my SD card. No idea what happened but reflashing fixed it.

nisargjhaveri, 2024-04-12:
From the conversation above, I see two issues.
1. Connection loop: it gets into some loop while trying to connect. This could be some mistake in the dongle side if it was working well before v0.4.0
2. Disconnection: sometimes the android auto stops after being connected. If this was not happening before v0.4.0, we might need to tweak some timeouts.

For the connection loop, think we need some more investigation into where it gets stuck. Logs should help in this case.

For disconnection also, logs might help, but my hunch is that the 10s timeout in tcp read might be too small and in some cases there might be a genuine delay more than that. We can increase the TCP read timeout as @Ioniq3 suggested above.

nisargjhaveri, 2024-04-12:
> Oh OK I see... No way to build an image that would dump logs to sd card? That way I could get logs properly after driving sessions.

/var/log is currently symlinked to /tmp, you can remove the symlink, and it should start storing the logs in sd card. I haven't tried this yes, but I believe this should just work.

For a long term solution, I was thinking about adding/removing this symlink based on some config. But for that to work, we need to read the config much earlier in boot. This needs some more changes. I have some local changes in that direction, but needs some more attention. I'll try to create a PR sometime next week.

Ioniq3, 2024-04-12:
To add more info, 

When it start to loop, i try to /etc/init.d/S92usb_gaget and S93 to restart and in this new implementation of the dongle from Commits from Mar 12, 2024 it doesn't work when in the old impletentation of the dongle feature it works perfect to avoid the physical reboot of the raspberry

OrdenKrieger, 2024-04-13:
I'm having the same issue since 0.4. Funnily I tested my pi zero 2 w first at my computer with Desktop Head Unit and there it would work just fine. In my car it would connect show aa on my screen and then disconnect until my car would show the device is unrecognized. It only helps to re-plug the device until it would eventually work. Might be that the board computer is much slower then running Desktop Head Unit on my computer, which causes the connection to timeout.

tiliarou, 2024-04-15:
> /var/log is currently symlinked to /tmp, you can remove the symlink, and it should start storing the logs in sd card. I haven't tried this yes, but I believe this should just work.
>
> For a long term [...]

Understood, I will try to remove the symlink.

I found an old fork of the AAwireless repo (I suppose it is before they launched the kickstarter), maybe there are some stuff in the 3 branches that can help : https://github.com/chutchi2/aawireless

tiliarou, 2024-04-19:
I believe I managed to log the loop, which ended up in a proper and valid connection by itself (after or connection/disconnection cycle, I never unplugged the pi)  
[10.0.0.1_20240419093849.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15037170/10.0.0.1_20240419093849.txt)
[10.0.0.1_20240419093910.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15037171/10.0.0.1_20240419093910.txt)

Ioniq3, 2024-04-19:
> I believe I managed to log the loop, which ended up in a proper and valid connection by itself (after or connection/disconnection cycle, I never unplugged the pi) [...]

Yes, I think , as I see in my log, is something related to "buildroot user.info aawgd[264]: Bluetooth device already connected, disconnecting"

nisargjhaveri, 2024-04-21:
Thanks for some logs @tiliarou.

The logs suggests that the phone repeatedly connects via BT to start the Android Auto when the dongle thinks it is already connected. I can see multiple "AA Wireless NewConnection" logs even when TCP is already connected and hasn't timed out.

Once TCP times out, we try to force reconnection with the phone and also see the "Bluetooth device already connected, disconnecting". Interestingly this log is seen even on first connection on boot, maybe the phone connected faster even before we tried connecting first? This should not be needed for the first time, but I don't think this should cause any major issues. For investigations, we can try removing the disconnection. For context, it was needed to actually force the phone to reconnect via BT in case the the connection fails for some reason. If we don't disconnect and reconnect BT, the phone never retries once successfully connected.

My hunch is that on the phone side, the phone was able to detect some problem in the TCP connection and is retrying, but on the RPi, we're still waiting for the timeout. One way to investigate more might be to kill existing session on "AA Wireless NewConnection" and restart the USB with fresh connection and see if that helps.

nisargjhaveri, 2024-04-21:
Also, does this loop also occur if you change the connection strategy to phone first? Does it connect more reliably in phone first, or instead of loop there are more failure in phone first with otherwise same setup?

tiliarou, 2024-04-21:
> Also, does this loop also occur if you change the connection strategy to phone first? Does it connect more reliably in phone first, or instead of loop there are more failure in phone first with otherwise same setup?

I have yet to try other modes, I remained with the default mode so far, I will try to change that.
But the dongle mode is the only one able to not fake HSP profile?

nisargjhaveri, 2024-04-21:
> But the dongle mode is the only one able to not fake HSP profile?

Yes, for dongle mode we don't need the fake HSP profile. For other modes we do need the fake HSP profile for connection. But in those cases, we disconnect BT as soon as the connection is established to make sure phone audio is routed to the headunit and not the dongle.

th873, 2024-04-22:
I have tried all modes and it is just dongle mode that is not working.

hkfuertes, 2024-05-07:
Don't know if it is the same issue, but I also had a bootloop when in dongle mode (well, maybe in all modes) that I get a "USB unrecognized" for 5-6 times in a row.
Setting this to higher (or infinite, 0) seems to solve the issue:
https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/9565add3d8abcc2cc378ac3b500e3dd28494ca9e/aa_wireless_dongle/package/aawg/src/proxyHandler.cpp#L172

I can try to create a pr to move this number to the .env file, if it helps...

manio, 2024-05-20:
> Yah, it was my SD card. No idea what happened but reflashing fixed it.

I had something similar. See [here](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/pull/109#issuecomment-2104929034) :)

ProtectiveManEgg, 2024-05-21:
> Don't know if it is the same issue, but I also had a bootloop when in dongle mode (well, maybe in all modes) that I get a "USB unrecognized" for 5-6 times in a row. Setting this to higher (or [...]

I just want to add on here; this did "fix" it for me, however still having seemingly random issues and intermittent issues with warm boots. 

I may have a different issue and if/when I get a chance to grab logs I will open my own. Using dongle-mode however makes my head unit's BT act up telling me to turn off BT and reconnect to the head unit (doesn't actually work), but it appears to work fine while it's saying that (Pi 0w2, Pixel 7 and 2020 Honda Civic).

ramon2k10, 2024-09-01:
Hi, I still suffer the same problem of connection loop. I understand that the problem persists and no solution has been found yet, ins’t it?