nisargjhaveri/WirelessAndroidAutoDongle #99: I cannot connect to the Bluetooth of Genesis.

DamienLee2017, 2024-04-03:
My car is a Genesis GV80. When I use Android Auto, I can't connect to the car's Bluetooth system.

"make": "GENESIS",
"model": "JX1",
"year": "2019",
The HUIG version of the HU is 1.5, and the Android Auto version of the phone is 1.7. Are there others who have encountered the same issue? Please help me, thank you!

Using AACS doesn't work, but there is no problem with the wired USB connection of Android Auto brought by the phone itself.

nisargjhaveri, 2024-04-03:
> Using WirelessAndroidAutoDongle doesn't work, but there is no problem with the wired USB connection of Android Auto brought by the phone itself.

What is the issue you're facing when using WirelessAndroidAutoDongle? What have you tried? How are trying to connect to the Bluetooth and how is it an issue?

closed 2024-04-07 by DamienLee2017

DamienLee2017, 2024-04-07:
Thank you for your attention, I'm sorry I remembered it wrong. It was AACS that had a problem, not the Wireless Android Auto Dongle.

> What is the issue you're facing when using WirelessAndroidAutoDongle? What have you tried? How are trying to connect to the Bluetooth and how is it an issue?

Thank you for your attention, I'm sorry I remembered it wrong. It was AACS that had a problem, not the Wireless Android Auto Dongle.