nisargjhaveri/WirelessAndroidAutoDongle #101: Rasberry pi 4 not conecting to head unit

bahmad6890, 2024-04-06:
I am trying to compare the performance between Raspberry Pi Zero and Raspberry Pi 4. I created an SD card with the pre-built image provided here. The Raspberry Pi Zero works perfectly, while my phone connects to the Raspberry Pi 4 and I can see my Wi-Fi is connected to the Raspberry Pi but not to my head unit, whereas the Raspberry Pi Zero is working fine. For the Raspberry Pi 4, I used a USB Type-A to Type-A cable and a separate 5-volt power supply. Could you advise on troubleshooting this issue?

posh6876, 2024-04-06:
I am also using rpi 4 and it is working fine.  However, i used the usb c not the usb a of rpi 4.  The version 4.0 works perfectly for my subaru ascent.

> I am trying to compare the performance between Raspberry Pi Zero and Raspberry Pi 4. I created an SD card with the pre-built image provided here. The Raspberry Pi Zero works perfectly, while my [...]

bahmad6890, 2024-04-07:
Thank you very much working fine now. with USB C

closed 2024-04-07 by bahmad6890