nisargjhaveri/WirelessAndroidAutoDongle #105: Retry Connection or Restart on Failed Android Auto Commands

NabeelUppel, 2024-04-12:
Hi! 

So I've been using the dongle and it works greats but sometimes it would either black screen on start up (it says connected but just a black screen) or after awhile  disconnect with usb not responding error. 

I couldn't get to the logs for the second error.

But it would be cool if it could it could restart automatically if commands fails or if display is blank (not sure if that's possible.)

I also want to know if there's a way to make logs persistent, maybe save it by date time and remove them every week or something. 

Also, would it be possible to SSH to it while my phone is connected to the wifi via Android Auto instead, so that way we can send commands to the RPi using Google Assistant.

admiralspeedy, 2024-04-12:
This sounds like a hardware issue related to the connection between the Pi and your head unit.

What kind of car are you driving/head unit are you using? Have you tried a different USB cable between the car and Pi, or a different USB port if your car has more than one?

As for SSHing from your phone, you should technically be able to by disconnecting from the AA Wireless network, then re-connecting. The SSH app might have to support choosing the network or you might have to also shut off mobile data so that the AA Wireless network is the only one you're connected to.

NabeelUppel, 2024-04-12:
Mazda 3 2015, the black screen seems to be related to Mazda, it doesn't happen  often like 1in 7 times. I just have to disconnect the usb and plug it back in again and it works. 
As for the disconnect, I'm not sure what cause that again it doesn't happen often but it would be nice to have an automatic recovery from these situations. I only have one plug and one cable available but other than these random time it doesn't seem to be an issues. 

So on my phone I have to WiFi networks available, the one "via Android Auto" and the one below that which I can connect to manually but I cannot do so because the Bluetooth will automatically reconnect to the "Via AA" one. 

Do you know if there's another way to send terminal commands to the RPi? I guess I could set up tasker to disconnect to the Bluetooth and try connecting to the other network.

admiralspeedy, 2024-04-12:
A new cable only costs a few dollars. I would definitely just buy one and see if that helps because the black screen and random disconnects with a USB error are almost certainly a bad cable or your head unit is faulty/runs terrible software.

There should only be one Wi-Fi network but thinking about it now, since 0.4 has connection recovery it would be hard to do since it will automatically reconnect. If you are seeing two networks can you post a screenshot?

The only easy way is with a PC or another phgone. This is why adding persistent SD card logs is being discussed because unless you carry a laptop/second phone with you or the issue happens when you are close enough to your house to connect to the wireless network with a desktop PC, it's impossible to grab the logs.

NabeelUppel, 2024-04-13:
I'm definitely gonna go get my hands on a new one, the issue is that all the ones I buy are mislabeled and aren't actually data cables but I'm gonna buy a good one soon. 

Yep two networks for some reason. 
[image]

Yep, it's not ideal to carry another pc/phone around just to check the logs, perhaps uploading to Google drive or something like that? But that might be more headache than it's worth.

posh6876, 2024-04-14:
Thanks to Nisarg, Version 0.4.0 works for me.  I am using rpi 4, usb a to c (otg), galaxy note 20 ultra (not rooted) and 2021 subaru ascent. 15sec max to complete the wireless connection. I kept "switch to better wifi-networks" to off.  I am seeing also 2 the same ssdi.

admiralspeedy, 2024-04-15:
> I'm definitely gonna go get my hands on a new one, the issue is that all the ones I buy are mislabeled and aren't actually data cables but I'm gonna buy a good one soon.
> 
> Yep two networks for some reason. ![Screenshot_20240412_213800_Settings](https://private-user-images.githubusercontent.com/52805376/322211291-2af7883e-6c4e-4bf4-bf0a-364ef572a06b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMxOTQ2NzEsIm5iZiI6MTcxMzE5NDM3MSwicGF0aCI6Ii81MjgwNTM3Ni8zMjIyMTEyOTEtMmFmNzg4M2UtNmM0ZS00YmY0LWJmMGEtMzY0ZWY1NzJhMDZiLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE1VDE1MTkzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ3NTA4NTgwOGVmMTZjMWIzY2U0Zjg1ZWU5OGI3M2FiZWJkN2RmZmQxYTBiMGY0OTYyODk4MDk5YmJmMDQ0YzcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.oJhwwTM1cIm7hSgY11cjHRJ3VJcL_gPnXxkWqvOhbVI)
> 
> Yep, it's not ideal to carry another pc/phone around just to check the logs, perhaps uploading to Google drive or something like that? But that might be more headache than it's worth.

I see people mention getting power-only cables all the time but I have purchased/ordered hundreds of cables in my life and I don't think I have ever got a mislabelled cable that only supplied power in my life. In fact, I have only ever even seen maybe 3-4 of them ever and they all came with devices that only use USB to charge and I always just kept them with those devices.

Where do you get your cables that this happens frequently?

nisargjhaveri, 2024-04-17:
The two wifi network showing up is normal and often happens if you also connect to the wifi network manually. It shows one entry for the manually connected network and one connected "via Android Auto". This should not cause any issue with the actual connection, and is handy when you want to get the logs out as you can switch to the manually connected one (though might be tricky with the retries).

NabeelUppel, 2024-04-19:
@admiralspeedy just whats available where I am I guess. But I ordered a good one and the same issue still persists. 
A black screen appears, not sure if its the code or if it's a head-unit issue.

nisargjhaveri, 2024-04-21:
For the black screen, does it not try to automatically reconnect in some time when this happens? In v0.4.0, we added some timeouts and reconnection logic to recover stuck connections. See if the logs helps identify what's going on?

NabeelUppel, 2024-04-21:
So from what I can tell, it's seems to be connected normally just a black screen. I can't be certain, when it does happen I'm never close enough to another device so I can pull the logs. 

I wonder if there's a way to have a sort of health check once connected but again this just be a my head unit issue rather than a software issue but hopefully next time it happens I can pull the logs to confirm

nisargjhaveri, 2024-09-19:
@NabeelUppel were you ever able to figure this out? Did any of the changes to retry connection help in this case?

naby-dev, 2024-09-19:
I think this either my phone or the headunit. Once this stopped working altogether I ended up using wired AA and it still occured. Feel free to close.

nisargjhaveri, 2024-09-19:
Okay, closing for now. Feel free to reopen if you get more ideas on how and if we can fix this on our end.

closed 2024-09-19 by nisargjhaveri