nisargjhaveri/WirelessAndroidAutoDongle #106: Hyundai Tucson Hybrid 2021 not working

tonydee2377, 2024-04-15:
Hi, 
I'm new here, bought a Tucson Hybrid Excellent latest model on the market released in 2021, and it works Android Auto via USB.
I wanted to enable this wireless function to avoid using the USB cable,
and I purchased a Rpi zero W 2017, followed the instructions, but unable to connect.
I tried to do something in the configuration and also on the phone, the result is that before it kept restarting, now it connects and Android Auto in the settings says "connected to usb" even if it is instead connected via WiFi on AAWirelessDongle.
Now if everything is ok, why doesn't my car give me the Android Auto icon, which instead appears as soon as I connect the USB cable to the phone?

Thanks for any help

tonydee2377, 2024-04-19:
I'll answer myself.
After a few tests it works.
I installed v0.3.1 first and it worked after a few minutes of waiting, then I reformatted the SD card well and installed the latest version v0.4.0 and this also worked.
I don't know what happened before, but having a few USB cables around, I found that many were fake. Now I continued to make customizations one at a time, until I discovered that changing the SSID and password no longer made anything work, but I also solved this by putting the right variables in the /etc/aawgd.env file, also tampering with the bluetooth file, but now everything works fine, I get into the car in time to turn on the navigator and Android auto connected via wifi.

Now having a 32Gb SD card and therefore some free space, I created a FAT32 partition on the remaining space in order to put mp3s and videos and use the zero rpi w also as an mp3 USB stick.

But it doesn't work, can anyone tell me how to show this partition on the USB cable to the car radio?
I understand that the USB is busy making the phone's Android Auto communicate with the car, but if I could switch to also use the space as an archive for my playlists, it wouldn't be bad.

I will try these days, in the meantime if anyone can help me I would be grateful.

nisargjhaveri, 2024-04-21:
Glad that you were able to make it working! :)

> Now having a 32Gb SD card and therefore some free space, I created a FAT32 partition on the remaining space in order to put mp3s and videos and use the zero rpi w also as an mp3 USB stick.
>
> But it doesn't work, can anyone tell me how to show this partition on the USB cable to the car radio?
I understand that the USB is busy making the phone's Android Auto communicate with the car, but if I could switch to also use the space as an archive for my playlists, it wouldn't be bad.

As you mentioned, I don't think it would be possible to use it as a usb stick at the same time that the Android Auto is also running. You might be able to switch between using it for Android Auto and USB storage using some customisation to gadget configuration we have. Though, not currently in the scope of this project. Let us know if you get it working though!

closed 2024-04-21 by nisargjhaveri

tonydee2377, 2024-05-30:
I tried to make the two things coexist, but it doesn't work.
Practically from Windows we can see that with AAWireless, the device is seen as an Android mobile device.

But by creating a new partition on the remaining 30GB and inserting a 30GB file system formatted in FAT32 and mounting it with:

/sbin/modprobe g_multi file=/media/piusb.bin stall=0 removable=1
mount -o ro /media/piusb.bin /media/usb_share

from this guide: 
https://www.instructables.com/Raspberry-Pi-Zero-W-As-WiFi-USB-Media-Device/
RPi is seen as a FAT32 mass storage drive, and I can copy my mp3s to it, the car reads them normally.

The problem lies in the driver perhaps, or you use modprobe -a dwc2 libcomposite, or modprobe g_multi for mass storage. Together there is no way.
Through the configuration of umtprd.conf I managed to create a Media drive seen under Android Auto in FAT32, but the problem is that Windows crashes as soon as I play an mp3, and the car doesn't see the mp3 files because they are not directly in a FAT32 drive, but under an Android device.

So for now, to have both things without unhooking the RPi from the car's USB port to insert a USB drive with music, I added a new service file S95button, which, through a small button connected to a gpio, allows me to switch from g_multi for music, or pressing again reloads the S92usb_gadget and S93aawgd services to bring AAWireless up again.

I don't know if in the future someone can add something that reads audio files on the SD card space.
Also solved USB instability problems, now it's fine.

Thank you so much nisargjhaveri