nisargjhaveri/WirelessAndroidAutoDongle #107: Device not Found error in Sony XAV-AX5000

john19922, 2024-04-20:
Hi, I'm a noob here but the project looks very interesting & I was trying to connect the dongle with my car head-unit Sony XAV-AX5000 but it shows Device not found. I'm using Raspberry Pi Zero W2 & v0.4.0 release. Another issue is when I'm connected to the 'AAWirelessDongle' Wifi my phone's interest stops as it thinks I'm connected to a regular internet wifi connection.

Attaching the error message..
[AAWireless Log.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15049223/AAWireless.Log.txt)

Thanks.

nisargjhaveri, 2024-04-21:
The logs seems to suggest your phone is not connected via Bluetooth to the RPi. You don't need to manually connect the AAWirelessDongle WiFi.

Steps to setup:
- Connect phone to Car via USB, make sure Android Auto starts
- Now connect RPi instead of the phone via USB with the Car.
- Connect phone to the RPi's bluetooth "AndroidAuto-Dongle". Make sure phone is also connected to the Car via bluetooth.
- Now it will automatically connect phone with the Wifi and start Android Auto.

nisargjhaveri, 2024-06-07:
Were you able to make it work by any chance?

closed 2024-08-26 by nisargjhaveri