nisargjhaveri/WirelessAndroidAutoDongle #108: Dacia Sandero III (2024)

hkfuertes, 2024-04-29:
Hello There... I had a Kia Niro, but yesterday I got a new Dacia Sandero... this is just to let you know that dongle mode works like charm on my RPI 3a+ and my Dacia Sandero (Media Display)...

PS: Hopefully I wont be needing this project any more, If I'm able to hack the infosystem and enable wifi. Upper models of Dacia DO have Wireless Android Auto.

hkfuertes, 2024-05-04:
... Although is there some kind of race condition on my phone when connecting to the info system and the dongle. It takes forever to connect to info system, and until that is not done, it won't initiate the native flow, so it loops like 5 times with errors until stable connection.

If I plug the pi once the bluetooth connection with the car is already established it connects on the spot with no errors...

So I will try to test something once I have some time... What if I remove dongle's Bluetooth from the equation?

My idea is to add another connection strategy (no bluetooth), and the proto part of the native flow, hardcore it into a launcher app that I will make. This app (with the manually imputed Wifi proto) will try to initiate android auto from within the phone, once this is connected to the info system.

Any thoughts or ideas? Is it feasible or makes sense? I will need to investigate how to connect to suggestions/discovery API for the wifi connection.... (Like android auto does)... Or launch android auto intent as the native flow does...

I'll keep you posted!

hkfuertes, 2024-05-04:
Oh... https://github.com/nisargjhaveri/AAWirelessGateway already solves the app part of my last comment! I will test it tomorrow then :)

nisargjhaveri, 2024-05-04:
If bluetooth connection to the headunit is slow and is causing the issue, how/why would removing dongle bluetooth connection help?

Also, curious, the phone first strategy should also help on your case, no?

And yes, the client part of that app should help in what you're trying to do!

hkfuertes, 2024-05-04:
It's something I need to test, but it seems that both Bluetooth fight for connection, making the connection with the info system take a lot longer to be established... If I don't use the pi, it connects fairly quicker than if the pi is present. 
That is what I wanted to test.

I also tested all connection strategies, and the only one promising is dongle mode, the rest either did not connect at all, or there are some issues.

I will try to run your Android app with the dongle as is (unpairing the bluetooth first)... If it's quicker I will try to implement the no bluetooth connection to make it cleaner.

I will keep you posted.

hkfuertes, 2024-05-06:
I was just reviewing the issues and maybe this is related to this #98 bug...

I was not able to make it work with the android app... it wouldn't connect. I want to test other strategy... how about `modprobe` from UsbManager? This would make sense when phone first, so that the headunit will only notice there is a device plug to the usb when there is a connection.

I will test this.

hkfuertes, 2024-05-07:
It seems that for now (I just built a CM4 version) the best option is to use Phone First. I will close this thread.
It is true that in the future, @nisargjhaveri, It would be cool to add full compatibility with your app. I could revamp it a little if you want and add a quick tile setting so that 2 persons using the same car, could decide which one connects to the dongle. I mean, leaving this responsibility to the users in the phone, rather than on the dongle.

I will try to `tee` aagw app to recover logs on why my dacia sais so many times (in all modes) "unrecognized usb" (it takes like 5 attempts).

Also, an before closing this issue, I just want to share with you this: https://github.com/tomasz-grobelny/AACS 
Maybe you knew already, but they reverse engineered the Android Auto protocol, and they even have a AAServer that "emulates" a phone.

Thanks for all the support! I will leave this open for if you want to reply, or you can even close this issue yourself.

nisargjhaveri, 2024-05-07:
@hkfuertes, Thanks for the investigations!

Yes, I've seen the AACS project, it's interesting. There is an underlying [aasdk](https://github.com/opencardev/aasdk) project as well which has low-level implementation for the AA protocol. It might be interesting if we can combine some of it to achieve things like #41 or #47. Though, it also can add a whole lot of new issues! We can maybe come up with some extension mechanism such that we can use something like AACS without having to copy/reimplement it here.

hkfuertes, 2024-05-07:
I think I found my issue, and think it might be related to #98 ... I explain it in my last comment. I will try to open a pr to move that parameter to the env file keeping the 10 secs as default.

Thank you for everything! @nisargjhaveri ... I close this issue.

closed 2024-05-07 by hkfuertes