nisargjhaveri/WirelessAndroidAutoDongle #110: error on connection and once connected, many loss (Audi 16 MIB2)

nicof38, 2024-05-05:
Hello,

Running the 0.4 version, the connection fails many time, seems not getting the data.
then once finally connected (3 or 4 loops of error), it gets connected, but without any errors visible in the messages log files, it is laggy, and sounds and controls are interrupted for a second or two, before resuming wihtout any actions.

Phone is Pixel 7 pro
[pi_zero_2w_messages.log](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15213300/pi_zero_2w_messages.log)

nisargjhaveri, 2024-05-07:
This one looks similar to #98. But I'm also seeing some usb errors I've not seen earlier, not really sure if they have anything to do with the issue, though.

Also, for the time being, you can try changing the connection strategy in /etc/aawgd.env and see if phone first strategy can give better results.

```
Jan  1 00:05:02 buildroot kern.info kernel: [  302.794436] dwc2 3f980000.usb: dwc2_hsotg_ep_sethalt(ep 9dea61b9 ep0, 0)
Jan  1 00:05:02 buildroot kern.warn kernel: [  302.796588] dwc2 3f980000.usb: dwc2_hsotg_ep_sethalt: can't clear halt on ep0
```

nicof38, 2024-05-07:
ok, did not saw first this one. Will read carefully it and check what can be done. Thanks

nicof38, 2024-05-08:
I have tested the connection strategy to "phone first" and it gets connected, but then after a few seconds, it drops everything and it goes again for a cycle of connection. I will make a dump of the logs tomorrow to have a better view of what is failing

nicof38, 2024-05-18:
OK, I've tried to power the rpi from both otg and power port. It stays on and connected for longer period, but still crash. Also, with phone first, when I loose network on my phone, it never resume the connection :( I have to restart the rpi.

Think I'm gonna try some pending PR with running my own build and see if any enhance the situation

nisargjhaveri, 2024-09-19:
@nicof38 can you check if this is still an issue with the latest version? I think most of this should be fixed now.