nisargjhaveri/WirelessAndroidAutoDongle #113: connection error after updating

mariolugo2, 2024-05-11:
connection error after updating

I updated to the latest version and I'm having trouble connecting.
pixel 7 pro 
mib2 gti 2017

log
[messages.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15281650/messages.txt)

nisargjhaveri, 2024-05-11:
Is your phone also connected to the the Car's bluetooth when this happens? For the new version, you need to once connect your phone with cable to the car, make sure Android Auto starts once, and then use the dongle.

mariolugo2, 2024-05-11:
> Is your phone also connected to the the Car's bluetooth when this happens? For the new version, you need to once connect your phone with cable to the car, make sure Android Auto starts once, and then use the dongle.

If correct, perform this process before using the dongle.

mariolugo2, 2024-05-14:
I still haven't solved the connection to the dongle, it sends me that it connects to Android Auto, then my phone screen flashes and a message appears about searching for Android Auto.

I already made the previous connection directly by cable.

I attach the logs that I extracted in the connection attempts

[image]
[image]
[image]

[messages.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15301737/messages.txt)
[messages22.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15301571/messages22.txt)
[messages2.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15301572/messages2.txt)
[messages222.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15301570/messages222.txt)

nisargjhaveri, 2024-05-16:
Some of your logs shows timeout while waiting for accessory start request. Might be related to #112.

mariolugo2, 2024-05-18:
> Some of your logs shows timeout while waiting for accessory start request. Might be related to #112.

Without success, I can't make the connection.
Add the timeout 10,30,60 and 0.

mariolugo2, 2024-05-27:
> Si desea experimentar más, ¿puede probar [la versión 0.3.1](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/releases/tag/v0.3.1) y ver si por casualidad eso ayuda?
> 
> Editar: me di cuenta de que comenzaste diciendo que tuviste el problema después de la actualización. ¿Puede confirmar que funcionó bien en la versión 0.3.1 y no en la versión 0.4.0 con la estrategia de teléfono primero? ¡Estoy empezando a darme cuenta de que algunas de las cosas que hicimos para mejorar la confiabilidad no siempre ayudan en algunas condiciones! :/

I already tried to go back to version 0.30, 0.31 and I have not been able to connect again.

I tried another phone, another vehicle, I changed the cable, SDcard memory.

The only thing I have left to check is to change the pi.

nisargjhaveri, 2024-08-25:
Were you ever able to resolve this issue?