nisargjhaveri/WirelessAndroidAutoDongle #116: Connection Issue: Honda Truck - RP zero 2 W

konradSC, 2024-05-18:
I'm having issues getting my Samsung Galaxy S24 connected to a Honda Ridgeline. I'm using a RP Zero 2 W with the latest build. 

I have attached the logs. 
[AA_logs.txt](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15365870/AA_logs.txt)

I tried everything I could think of with no success. 

Thanks for your assistance!

jeffhoagland, 2024-05-18:
I read through your log file and your issue and my issue seem similar to me.

In my case, BT connects, but then appears to simply stop.  In my BT interface menu on my ZFold5, the BT object for Android Auto appears in the list on my phone, but there is no options showing for me to select or connect to it, and it never asks for me to work through the security steps to get connected

[dnsmasq.leases-20240504.docx](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15365978/dnsmasq.leases-20240504.docx)
[hostapd-20240504.docx](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15365979/hostapd-20240504.docx)
[Android auto log 2 20240504.docx](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15365980/Android.auto.log.2.20240504.docx)
[Android Auto log 20240504.docx](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/files/15365981/Android.Auto.log.20240504.docx)

I made a couple different attempts as well as inspected the dnsmasq file to see what was going on and hopefully figure something out, but I was not successful.

I'm going to also submit my own issue in here, and see if my situation is the same or if the developers say it's different

konradSC, 2024-05-20:
Everything works using a RP4, so this issue seems related to the RP Zero 2W. @jeffhoagland

vishnu3095, 2024-09-04:
i have a similar issue with pi zero 2 . i added a second usb cable to the power input and that seem to fix the issue.

nisargjhaveri, 2024-09-19:
#88 has more info when a second usb cable is also required.

Apart from that, theatest version fixes some more issues and I believe this should already be fixed.

Closing this issue for now. Feel free to reopen as needed.

closed 2024-09-19 by nisargjhaveri