nisargjhaveri/WirelessAndroidAutoDongle #118: Disconnection problem on Renault Zoe / R-Link1

manio, 2024-05-20:
Hi
I have Renault Zoe from 2016 with R-Link1 (or maybe evolution, not sure), [pic here](https://www.gps-rlink.com/r-link-1/)
I am using Raspberry Pi Zero 2 W.
The problem is that from time to time I have AA disconnection. I know that it may occur, but the main problem is that the `aawg` is in state when it is not trying to recover. I cannot see the AndroidAuto icon on my head unit until I reconnect the dongle (power cycle) or it is also working when I log in into the dongle via `ssh` and manually restart the daemon:
`/etc/init.d/S93aawgd restart`
Then it is working again.

Full logs I've collected for this problem:
https://skyboo.net/temp/aa/10.0.0.1_20240427150919.txt
https://skyboo.net/temp/aa/10.0.0.1_20240506054835.txt
https://skyboo.net/temp/aa/10.0.0.1_20240506175836.txt

I was reporting this in https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/34, but I think that it will be better to create a new issue for this as it seems it may be quite different issue.

Here is what I've tried so far to solve this and doesn't help:
- [x] changed connection strategy to 1
- [x] changed connection strategy to 2
- [x] changed usbaccesory [timeout to 0](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/34#issuecomment-2100086908)

What I am now planning now is to try the @toek79 [branch](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/compare/main...toek79:WirelessAndroidAutoDongle:main)...

nisargjhaveri, 2024-05-20:
Your last log suggests a successful connection, was it established?

The other two, yes, looks like aawg is stuck and not timing out from an earlier TCP connection. As I suggested in the other thread, we can try to force close the earlier connection on a new bluetooth connection request and see if it helps with the loop.

nisargjhaveri, 2024-05-20:
#98 might also be interesting. Also, what happens when using phone first connection strategy? Would you be able to post logs from that as well?

manio, 2024-05-20:
> Your last log suggests a successful connection, was it established?

nope - If i remember correctly there was nothing in the log and the AA icon was gone and I cannot restore it. Regarding the log: at about `Jan  1 00:01:06` it stopped working so about minute later waiting, I manually disconnected and connected to the AP via ssh to obtain the logs.

> Also, what happens when using phone first connection strategy? Would you be able to post logs from that as well?

That mode just doesn't help with the issue (disconnected and cannot reconnect until restart) so when I saw that the problem still persist (cannot recover) I just didn't obtained the logs. Another problem with this mode was that my phone was exposing android auto icon even when I was close to the car and the dongle was not connected to the headunit USB, so for instance I cannot use home WiFi because it was reserved (but it is normal in that mode I think, as the name said).

Regarding #98 Thanks, I'll look into this but please let me check @toek79's changes first. I have it prepared to store on the SD and test again.

My main problem is that I am not using it daily and I didn't discover any way to trigger the bug easily. I just have to use it for some time to trigger the disconnection problem, but when it happens it just cannot recover the connection.

manio, 2024-05-26:
Hi,
Today I also have this problem. It is on recent build with https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/commit/f5965d9f8c1306562496b16f53cbcb8521b30b95.
The problem is as usual: I have stuttering sound from Spotify, then it disconnect and cannot reconnect again.
I disabled bluetooth, connected with WiFi and got this logs:
https://skyboo.net/temp/aa/10.0.0.1_20240526094117.txt
What should I try now?

sirferl, 2024-05-29:
Hi, thank you for your work.
I have an Renault Megane with the Rlink 1, too.
I tried the Rev 0.4 on a zero 2w: I could pair phone, but phone never connected to bluetooth, with or without USB connection to car. ( repeated entry in messsages: device already connected, disconnecting)
I then tried out rev 0.3.1 and it worked immediately.

manio, 2024-06-06:
@nisargjhaveri 
As I said I am now using strategy=1 (phone first) and I have to say that I have a feeling that the AA icon is available much faster comparing to default strategy=0. I was able to collect this kind of logs so far in this mode:

First scenario: I started charging the car, so the dongle was powered ON for over 2 hours from boot. My phone discovered the AA, don't know about the car because I just started charging the car and went away. After about 2 hours I just stopped charging, sit inside the car with my phone and started driving. Unfortunately the AA icon was not showing on my car head unit, so at `Jan  1 02:40:10` I connected to the dongle's AP and collected the log:
https://skyboo.net/temp/aa/10.0.0.1_20240605131252.txt

Second scenario: After a clean dongle start I was using the AA without any problem for about 4 minutes. Then I leaved the car with my phone in pocket. It was disassociated:
```Jan  1 00:04:12 buildroot daemon.info hostapd: wlan0: STA 82:86:99:74:a8:e5 IEEE 802.11: disassociated```
Then after a short while (maybe another 3-4 minutes) I returned to the car but the AA icon was not showing again. I leaved it longer waiting to appear. Unfortunately it was not, at `Jan  1 00:13:20` I connected to the AP to collect the logs:
https://skyboo.net/temp/aa/10.0.0.1_20240605133601.txt

nisargjhaveri, 2024-06-07:
For the first scenario, the following line in the log seems interesting, and it stops trying bluetooth connection after this. (maybe crashed?)
```
Jan  1 00:41:26 buildroot daemon.warn dbus-daemon[159]: [system] Connection ":1.2" is not allowed to add more match rules (increase limits in configuration file if required; max_match_rules_per_connection=512)
```

The second scenario is puzzling. TCP should've timed out after the wifi disconnection at some point and should've retried the connection. The logs are not giving any more hints then this. Does this happen frequently? You can maybe try with some additional logs to understand what might be happening.

manio, 2024-06-08:
> For the first scenario, the following line in the log seems interesting, and it stops trying bluetooth connection after this. (maybe crashed?)

Maybe - next time I'll do a `ps` to check this. I am sometimes stopping the `aawgd` manually - otherwise my Android is disconnecting WiFi, so I cannot make a reliable ssh - but this is all **after** the main problem.

> The second scenario is puzzling. TCP should've timed out after the wifi disconnection at some point and should've retried the connection. The logs are not giving any more hints then this. Does this happen frequently?

Frankly speaking, I have to say that it "mainly" works quite right. But you know - I am noticing problems and I had a situation when I needed the navigation the most because I was driving longer and I had to reboot because even toggling the AirPlane/Offline mode on my android didn't help. I will try to make some reproduction pattern with this if I could.

> You can maybe try with some additional logs to understand what might be happening.

I'll be glad - is there some verbosity switch in the `aawgd`?

manio, 2024-06-13:
update:
I was not able to reproduce the problem - it seems this is a random issue. But I've collected something different...

On clean start I had a problem connecting to the car. I was waiting about 2-3 minutes before I've got the AA icon in the car.
Here is the log: https://skyboo.net/temp/aa/10.0.0.1_20240611055525.txt

KadenK, 2024-06-22:
I've been having a very similar problem on my rpi02w as well, but my rpi4 has not had any issue at all. Admittedly, I have very little experience in low-level/embedded systems, but it seems like the rpi02w has a configuration issue in the defconfig mixing the 32-bit and the 64-bit. When I built and ran with the changes, it seems to be working just fine.  I made a PR if you want it, or you can just make the change yourself. #129

nisargjhaveri, 2024-06-27:
> On clean start I had a problem connecting to the car. I was waiting about 2-3 minutes before I've got the AA icon in the car.
Here is the log: https://skyboo.net/temp/aa/10.0.0.1_20240611055525.txt

The logs suggests that the bluetooth connection sequence did complete properly multiple times, but the TCP connection was never made, until it was connected after a few retries. Unfortunately, it is difficult to figure out why this happened just from the logs.

One possibility is that the phone was slow was going to connect but we ended up retrying before the phone had a chance to connect using tcp. I see we're retrying only after ~15-20s. But if this happens frequently for you, you can try increasing the retry timeout and see if that helps.