nisargjhaveri/WirelessAndroidAutoDongle #122: DSP/EQ

Larry3best, 2024-06-01:
Great project, but is there a way that DSP or an equalizer can be added? My car's eq is very limited. Thanks

nisargjhaveri, 2024-06-06:
This project sits between the phone and the headunit. And the DSP/EQ needs to be between the headunit and the speakers as I understand? I might be wrong, let me know if you have more ideas on how we can do this. 

If at all you can approximate EQ, you might have more success with some app on the phone itself.

nisargjhaveri, 2024-06-07:
Closing this as out of scope for now. Feel free to reopen if you have more thoughts on how and why this can be incorporated.

closed 2024-06-07 by nisargjhaveri

Larry3best, 2024-06-14:
Thanks for responding, i opened this because of the inconsistent EQ activation of JamesDSP and viper4android when using android Auto. But then I found that this project fixed my issues. https://github.com/WSTxda/ViperFX-RE-Releases