nisargjhaveri/WirelessAndroidAutoDongle #124: USB not supported

M-Teghtmeyer, 2024-06-05:
I am having problems connecting my Raspberry Pi Zero 2 W to my Sony head unit XAV-AX3000. I am getting the error USB device not supported. I did not see the wifi or the bluetooth signal for the device  either. 

Thanks for your help

nisargjhaveri, 2024-06-06:
It would be useful if you can share some logs.

nisargjhaveri, 2024-06-07:
You need to ssh into the Pi using your phone or some other device and get the logs. [The readme](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle?tab=readme-ov-file#troubleshoot) contains some information on how you can connect to the Pi.

pinozero21, 2024-08-25:
@M-Teghtmeyer how did you flash the image to the SD card? Because I used the pi imager tool and didn't work, same behavior as you described. But, after using balenaetcher tool, it worked and I was able to see the Bluetooth and wifi SSID. Hope this helps.

M-Teghtmeyer, 2024-08-25:
Thanks I will definitely try that. That makes a lot of sense it felt like it never installed properl

M-Teghtmeyer, 2024-08-25:
I got it to work. I think it was a combination of many things. But you made me want to try again. Thanks for your help. @pinozero21

nisargjhaveri, 2024-08-26:
Glad that you were able to make it work. If possible, can you please share what was the issue and how you resolved? It may help others. :)

closed 2024-08-26 by nisargjhaveri

M-Teghtmeyer, 2024-08-26:
I missed the prebuilt images the first couple of times fiddling around with it on the website . I did have a image built already from last time but and had tried to mount it a couple of times but never seemed to work correctly (not sure if i had the right raspberry pi) . I believe I had originally tried from the Docker option. I couldn't get the wireless AA dongle to show up or the Bluetooth option and my head unit told me the device wasn't compatible.

Sukesh2206, 2024-08-28:
Hi There
Could you please help me , I am facing same issue with my Sony xav5500. I am able to connect to wifi and Bluetooth but its showing g USB not found.

nisargjhaveri, 2024-08-28:
Please check your usb cable and the usb port you're using on the board. README has some more info. Feel free to create a new issue with logs if it still doesn't work.