nisargjhaveri/WirelessAndroidAutoDongle #126: Not connecting "Looking for Android Auto" -- Pi 3 A+  -- Ford Sync 3 (Works on Pi 4)

anthonyg45157, 2024-06-09:
First off just wanna say hats off for the project! Its great to see this be possible with the Raspberry Pi.

For some reason I cannot get the Pi 3 A+ to work using v0.0.4 or the previous build (v0.3.1) Everything works as expected up until the moment android auto should be popping up on the head unit.  I see a message that pops up on my phone and endlessly says "Looking for Android Auto.  This is with a brand new Pi 3A+ paired with A One Plus 12 on Android 14.  To take it a step further I cannot get the Pi 3 A+ to work on a second vehicle (2023 Subaru) it produces the same issue. 

This works exactly as expected with a Pi 4 using the exact same steps for both vehicles.   I have tried uninstall android auto, clearing data, factory resetting the vehicle. I've also tried multiple USB cords. 

Once again, it seems Bluetooth and Wifi are connecting perfectly fine but after the wifi connects Android Auto will not open. 

Any ideas or recommendations would be great. I can definitely get logs, I just need a little help with how exactly to go about doing that.

 

[image]

[image]

anthonyg45157, 2024-06-10:
Got a log file today , tried a few new cables as well. No luck 

```
# cd var
-sh: cd: can't cd to var: No such file or directory
# cd /var
# cd log
# cat messages
Jan  1 00:00:02 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:02 buildroot user.info console: Starting syslogd: OK
Jan  1 00:00:02 buildroot user.info console: Starting klogd: OK
Jan  1 00:00:02 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-03-30 17:19:50 UTC)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:02 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61-v7 (runner@fv-az1106-856) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP Sat Mar 30 17:50:47 UTC 2024
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model A Plus Rev 1.1
Jan  1 00:00:02 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]   Normal   empty
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] percpu: Embedded 12 pages/cpu s18004 r8192 d22956 u49152
Jan  1 00:00:02 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s18004 r8192 d22956 u49152 alloc=12*4096
Jan  1 00:00:02 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:02 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  root=
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Memory: 337220K/421888K available (9216K kernel code, 718K rwdata, 2316K rodata, 1024K init, 520K bss, 19132K reserved, 65536K cma-reserved)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]      Tracing variant of Tasks RCU enabled.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000018] Switching to timer-based delay loop, resolution 52ns
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000355] Console: colour dummy device 80x30
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000866] printk: console [tty1] enabled
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000917] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000958] CPU: Testing write buffer coherency: ok
Jan  1 00:00:02 buildroot kern.info kernel: [    0.001016] pid_max: default: 32768 minimum: 301
[... cut ...]
Jan  1 00:00:02 buildroot kern.info kernel: [    0.597944] EXT4-fs (mmcblk0p2): orphan cleanup on readonly fs
Jan  1 00:00:02 buildroot kern.info kernel: [    0.600399] EXT4-fs (mmcblk0p2): recovery complete
Jan  1 00:00:02 buildroot kern.info kernel: [    0.604587] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Quota mode: none.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.609166] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.612194] devtmpfs: mounted

```

dvelabs, 2024-06-15:
Had the same issue and my problem was that i had plugged the usb to the wrong port. 

Others had it with a failing usb cable.

anthonyg45157, 2024-06-15:
> Had the same issue and my problem was that i had plugged the usb to the wrong port.
> 
> Others had it with a failing usb cable.

Can you be more precise? Wrong USB on car or the board itself? I have tried both ports on my car with no luck. I am using only the micro USB port, do I need to be using another?

I've tried 5-20 USB cords that I know are data compatible as I've transferred data with them using a PC.

Appreciate the assistance

nisargjhaveri, 2024-06-15:
You need to use the USB-A port on the Raspberry Pi 3A+, with a USB-A to USB-A cabel as mentioned in #11. I believe the microusb cabel is only for power in this version.

anthonyg45157, 2024-06-15:
> You need to use the USB-A port on the Raspberry Pi 3A+, with a USB-A to USB-A cabel as mentioned in #11. I believe the microusb cabel is only for power in this version.

This would definitely explain why I'm having an issue. 

So for the pi 3A+ I will need 2 wires? 

1 USB A to micro USB and 1 USBA male to USB A male , correct,

anthonyg45157, 2024-06-15:
Ohhhhh wow, you can power the PI A+ using USB A to USB A very neat! Working as expected now!

anthonyg45157, 2024-06-15:
See above 👍

closed 2024-06-15 by anthonyg45157