nisargjhaveri/WirelessAndroidAutoDongle #136: laggy rpi zero 2 w

ccebrecos, 2024-07-17:
Tested the image with the Pi4 and works perfectly. But the image for the rpi zero 2 w it's laggy making it non functional. 

Is there anything I can do to improve the performance?

lob-star, 2024-07-24:
Same. Mine was very hot and eventually started to boot loop faster and faster.

nisargjhaveri, 2024-07-29:
Yeah, multiple people have reported similar issues with RPi 2 W. Someone with the issues need to debug what's going on. #66 has some more discussion around an similar issue.

ccebrecos, 2024-07-30:
Tested with another car, this time is a Volkswagen Polo and works by far better.
The first car was a Volkswagen Scirocco, newer by a couple of years, I was expecting to work better on a newer car.

So, I'd say it's something related to the car, maybe the power supply?

lob-star, 2024-07-30:
> Tested with another car, this time is a Volkswagen Polo and works by far better. The first car was a Volkswagen Scirocco, newer by a couple of years, I was expecting to work better on a newer car.

I'm running mine on a 2017 VW Golf R. It has the MIB 2.0 system. https://imgur.com/2tKXyQ6

cardassianscot, 2024-07-31:
If it helps, 0.31 didn't have the same issues. It had more issues connecting but once connected it was rock solid stable. 0.40 connected much more reliably but it's often virtually unusable on my pi zero 2 w. I can cope with the lag, it's the choppy audio that makes it unusable. So far using a pi 4 seems much better.
(Using a mg5 EV)

ccebrecos, 2024-08-02:
I tested 0.31 and works way more better with the same setup, without extreme lag, so now it's usable!

nisargjhaveri, 2024-08-02:
This is very curious. Can you try latest main branch version? I'd like to pinpoint if the reason is because of the use of "dongle mode" or some other change that went in between? It'd help a lot if we can identify that.

ccebrecos, 2024-08-02:
Sure. I'll give it a try and I'll let you know

cardassianscot, 2024-08-02:
Sorry not in a position to compile this at the moment, but happy to try a prebuilt version.

nisargjhaveri, 2024-08-03:
You can use rha artifacts from https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/actions/runs/9421948282

Antho91, 2024-08-03:
I'm flashing it and will test it on my 1 hour drive :) (today)

ccebrecos, 2024-08-03:
The image seems to not be properly configured:

[image]

[image]

Antho91, 2024-08-03:
On Windows with raspberry img burner tool it flashes just fine

Antho91, 2024-08-03:
Just drove for an hour and i could feel no lag.

Touch was fine and audio also.

Was using Maps and Spotify

Antho91, 2024-08-03:
3the test and laggy. Dont know what has been changed 🤷 

Laggy interface and laggy audio

ccebrecos, 2024-08-04:
Tested a couple of times; way more fluid than the v.0.4.0 but still laggy. Some chunks get lost and spotify is not fluid. Still usable, but not fluid enough to finish a song without any jump. 

Second time had a better experience than the first. 

May the temperature be related?

nisargjhaveri, 2024-08-04:
Okay, this is getting even more interesting. I went through the changes since 0.3.1 and nothing apparent is coming to mind that can cause this. If you have consistent lag in newer versions and no lag 0.3.1 consistently, you can try bisecting to figure out the change that causes this. This may require multiple trials and multiple builds though, I don't think all the builds would still be available in Github Actions. https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/compare/v0.3.1...v0.4.0 should show all the changes between the two versions.

cardassianscot, 2024-08-07:
Tried this version, for a couple of hours long journeys and it seems to work fine. There may be a slight lag compared to a cable but it is useable and there is no choppy audio.

nisargjhaveri, 2024-08-08:
Glad that the latest version seems to be better. I'll create a new release.

Btw, between 0.4.0 had the dongle mode as default. Both 0.3.1 and the latest main have phone-first connection as default. Maybe that is messing something and causing the lag?