nisargjhaveri/WirelessAndroidAutoDongle #137: Fiat Cronos S-Design 2021 w/ Pi Zero 2w requires 2 usbs

mvincenzi96, 2024-08-09:
Hey everyone.

Just wanted to comment that this wonderful project does work in my car, a Fiat Cronos S-Design 2021.
However, it's the same scenario as descripted in #88, which makes it significantly inconvenient to use: I need to plug in two usbs cables in an specific order for it to work.
And this car has only two usb outputs, rendering it uncomfortable for daily usage.
You first gotta plug in a cable for the power input, then plug in the otg cable, and it will work. 
Yeah, you can unplug the power cable once Android Auto mode kicks in. Nevertheless, the moment you stop the car, go do stuff, turn back to the car and start the engine, it won't automatically kick in Android Auto mode. You gotta unplug the otg cable, plug in the power, then plug in again the otg cable for wireless Android Auto to start.

In case you wonder, I am using a 6A data cable for the otg connection, yet it seems to not be enough to feed the Pi Zero 2 via otg.

mvincenzi96, 2024-08-26:
Same thing happens with Raspberry Pi Zero W, it requires two usbs

nisargjhaveri, 2024-08-26:
Let's discuss in #88. Feel free to add more info there if you find anything.

closed 2024-08-26 by nisargjhaveri