nisargjhaveri/WirelessAndroidAutoDongle #138: PI SD Card

borsosgy, 2024-08-14:
hello nisargjhaveri
I use the image with PI ZERO 2W.
The PI is powered via USB.
If I turn off the car, the power to the PI is switched off.
Because of this power outage, the SD image is broken every month and I have to create it again.
Do you have a suggestion as to what kind of SD I could use?
Thanks in advance

borsosgy, 2024-08-15:
can this help or work?

http://comfilewiki.co.kr/en/doku.php?id=comfilepi:read-only_file_system_with_overlay

Thanx

EgorkillerUA, 2024-08-20:
I can also confirm this. Only for me it breaks even more often, about once every day or two, which makes it impossible to use.

nisargjhaveri, 2024-08-20:
#109 has some details and discussion on how to prevent it.

borsosgy, 2024-08-21:
> #109 has some details and discussion on how to prevent it.

Hi nisarghaveri,

do you mean log2ram?

nisargjhaveri, 2024-08-24:
We already log to RAM, the logs are not written on the SD card. But the SD card itself is not made readonly as we do need to write the bluetooth pairing information occasionally.

#109 contains some instructions on making the SD card readonly after initially pairing bluetooth, which might help if your SD card is getting corrupt frequently.

sud007, 2024-09-12:
I too am facing this same issue with my Suzuki car. I have a Samsung SDXC card. I have to reformat and flash it every time, and then it works.

It breaks every day on first turnoff. I think this happens when music was playing and car was turned off.

sud007, 2024-09-12:
> We already log to RAM, the logs are not written on the SD card. But the SD card itself is not made readonly as we do need to write the bluetooth pairing information occasionally.
>
> #109 contains some [...]

When are we planning to merge this PR?

nisargjhaveri, 2024-09-12:
@sud007 you can try out builds from #159 and see if it helps? I'm waiting for someone try out and see if it actually helps.

sud007, 2024-09-21:
OK so thanks @nisargjhaveri for replying. Here is one thing that I did as mentioned on #109 

> edit the `cmdline.txt` file and add `ro` at the end of the line

1. I edited the file as mentioned and nothing else.
2. No image code building, just added `ro` at the end of command

And have tested this to be working for last 9 days (since my last comment).
What is working

1. Perfectly connects to my S24 Ultra.
2. Also added other devices without doing anything after the changes made in `cmdline.txt`. (Pixel 6 and S23 Ultra).

Been 9 days since no issues!

nisargjhaveri, 2025-03-28:
The corruptions issues should be fixed in the latest release v0.5.0 via #159.

closed 2025-03-28 by nisargjhaveri

sud007, 2025-03-29:
> The corruptions issues should be fixed in the latest release v0.5.0 via [#159](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/pull/159).

Awesome t! Thanks for the update buddy.