nisargjhaveri/WirelessAndroidAutoDongle #140: Will it work with RBP 3 B+

Anupamp-009, 2024-08-21:
Hi,

I have booted my RBP 3B+ with the 3A image. So my question is my RBP has 4 USB port, and I don't know which USB port it will work with. can I use any of the USB port or I have to make any changes?

nisargjhaveri, 2024-08-21:
Raspberry Pi 3 B+ is not supported as it does not have any USB OTG enabled port.

closed 2024-08-21 by nisargjhaveri

Anupamp-009, 2024-08-22:
Thanks for your quick response. I read in an article that 

> You cannot use the Raspberry Pi 3 as a USB gadget. At the time of writing this is only possible on the Raspberry Pi 0, 0W, A, A+, and 4.   The Pi B, B+, 2, 3, 3A+, and 3B+ lack the required pin for OTG.

So if I am going to buy a 3A+ from the below link then I need you confirmation. Also which port supports OTG?

[https://www.thingbits.in/products/raspberry-pi-3-model-a?taxon_id=62](url)