nisargjhaveri/WirelessAndroidAutoDongle #142: Images are corrupt?

arrees, 2024-08-22:
Hi, When downloading PI 3A the image and then trying to write it, I get errors. I've downloaded it multiple time with the same issue.
 
[image]

I've tried Etcher, Rufus etc. Using Zzip for Extracting the img shows an corruption error too.

Cheers,
Andy

CsabaLevente, 2024-08-22:
same here with raspberrypi0w-sdcard.img.xz

superbillen, 2024-08-22:
You have to extract the file with 7-zip its still compress as the Xz extension are ind the end.
When its extracted and only.img you can flash this

nisargjhaveri, 2024-08-22:
Looks like an issue with Balena Etcher. See https://github.com/balena-io/etcher/issues/4064, https://github.com/balena-io/etcher/issues/4258. Seems like after uncompressing, it might work.

You can also use alternate softwares such as [Raspberry Pi Imager](https://www.raspberrypi.com/software/) also in the meantime.

arrees, 2024-08-22:
Yes, you're correct. Raspberry Pi manager works! THank you.

Should the pi3A+ image work with a pi3B+?

nisargjhaveri, 2024-08-22:
RPi 3 B+ is not supported. See #140

closed 2024-08-22 by nisargjhaveri