nisargjhaveri/WirelessAndroidAutoDongle #143: 2020 Chevy Silverado doesn't find AA Wireless devices / can't SSH

moose09876, 2024-08-22:
Vehicle is a 2020 Silverado. Wired AA works great. Multiple dongles will not show up as AA Devices. Stumbled here to try and get some logs / troubleshooting, but after the device boots and I connect with my phone, I can't SSH to the device from my laptop. It just throws a a connection refused error, never prompts for creds.

nisargjhaveri, 2024-08-24:
The logs does not seem to clearly point to a an issue. Both USB and Wifi connection seem to be started correctly. A quick search seems to suggest more people might be having similar issues with other adapters as well, and unpairing and clearing cache of both headunit and phone seems to help sometimes. I can't confirm though, but maybe worth a try?

moose09876, 2024-08-24:
I've had 3 different phones since I've started trying this. The phone side works fine. The truck does not see any of the AA Wireless device as Android Auto compatible, including the Pi.

moose09876, 2024-08-26:
Trying to find out how to get some more logging to see what the difference is between the phone and the wireless adapter, but not finding much.

[image]
[image]
[image]

moose09876, 2024-08-28:
Thanks to Motorola support for giving me some steps that resolved this:

- remove the AAWireless device from the phone's Bluetooth devices
- remove the head unit from the phone's bluetooth devices
- remove the phone from the head unit's list of Bluetooth devices
- reset the AAWireless device (other devices, press and hold the button on it for about 45 seconds; during that time it should change color several times
- unplug the AAWireless device
- connect AAWireless device to vehicle USB
- pair AAWireless device with phone in the phone's bluetooth settings
- wait for 30 seconds while phone connects
- after the Map screen/Android Auto should pop up
- phone should prompt for bluetooth access to vehicle

closed 2024-08-28 by moose09876

nisargjhaveri, 2024-08-28:
Glad that you were able to make it work. Thanks for sharing the steps as well.