nisargjhaveri/WirelessAndroidAutoDongle #144: Toyota Urban Cruiser / Suzuki Brezza not work

rolfbartels, 2024-08-24:
I have tried with a rPI4 and rPI3A+

I am able to get the phone to connect to the WAAD but the WAAD does not connect to the car.

moose09876, 2024-08-24:
This sounds exactly like mine, see previous issue. I pulled logs, waiting for response.

nisargjhaveri, 2024-08-24:
The most common issue when the phone connects but AA does not start is faulty USB cable. You can try to get the logs to get more info on what might be going on.

hitensam, 2024-08-25:
Hi, I tried using the Raspberry Pi Zero W. It didn't work for me—am I doing something wrong @nisargjhaveri ?
- I can see it in the Wi-Fi list.
- I made sure to use a micro-USB data cable.
- I plugged it into a port that supports OTG.

arnaullv, 2024-08-25:
Make sure you are using a quality data microusb cable, and not one from aliexpress...

rolfbartels, 2024-08-25:
@nisargjhaveri 
would this be the same for Pi Zero ?
I hae uploaded my pi3a+ logs you can take a look, but I didn't use a USB-A to USB-A cable,

[rpi3aplus.txt](https://github.com/user-attachments/files/16741336/rpi3aplus.txt)

nisargjhaveri, 2024-08-25:
The micro-usb port on Pi 3A+ does not support data transfer at all.

For RPi Zero W and Zero 2 W, you need to use the micro-usb port marked as "USB" and not the power one.

rolfbartels, 2024-08-25:
Awesome thanks

hitensam, 2024-08-25:
It worked!
Thanks, @nisargjhaveri, for creating this.
This solution should definitely work for all Maruti and Toyota cars with screens like this.

[image]

nisargjhaveri, 2024-08-26:
@rolfbartels RPi 3A+ with USB-A to USB-A should work. I'm closing this issue for now. Feel free to reopen if it still does not work.