nisargjhaveri/WirelessAndroidAutoDongle #146: Not able to connect using raspberry pi 0W but working fine with raspberry pi 4

gaurishpathare, 2024-08-24:
I am using the latest version 0.4.1 for my seltos. I tired with raspberry pi 4 and it worked fine.  But when I tried to do the same using my raspberry pi 0 W as pi 4 takes lot of space out does not work. Can you please suggest why that could be ?

nisargjhaveri, 2024-08-24:
If RPi 4 is working and RPi Zero W is not, one of the most common issue can be the USB cable you're using or the USB port you're using in the RPi Zero W. Make sure you're using the OTG enabled USB port with a good quality data cable.

If that does not work, please provide some more info on where it gets stuck and provide logs if possible. That'll help finding out the issue.

gaurishpathare, 2024-08-25:
Hi @nisargjhaveri , thanks for the response. 
Providing some additional details here. I have attached the logs for both pi4 and pi0 ( where its not working) . 
[messages.NotWorkingPiZero.log](https://github.com/user-attachments/files/16738502/messages.NotWorkingPiZero.log)
[messages.WorkingPi4.log](https://github.com/user-attachments/files/16738503/messages.WorkingPi4.log)

nisargjhaveri, 2024-08-25:
The logs also suggests issues with usb. The wifi side if connection is working well, but there is no communication happening on the usb. Since the headunit is working file with RPi 4, we can rule out an issue on the headunit side.

gaurishpathare, 2024-08-25:
I did try with multiple usb cables all of which are data cables. Also it's a brand new raspi 0W which i purchased. Can you suggest what can be tried to fix the connectivity?