nisargjhaveri/WirelessAndroidAutoDongle #150: 2021 Lexus Rx350

rednex269, 2024-08-25:
I have a factory head unit on a 2021 Lexus Rx350. I am using a Pi Zero W . I was able to get connected to bluetooth and then to wifi... the phone acts like it is connected, however on the head unit, the "projection" option never changes to Android Auto, where it will allow it to show.  I am having issues getting connected to the device because Idon't have hdmi handy..   I don't have a ethernet adapter and would have to connect to it via wifi,and I"m sure that will affect the setup of the device?

Thanks VERY  much for the assistance!

Joey Sevy

bradhmn, 2024-08-26:
I think I have the same/very similar problem with my 2017 Hyundai Santa Fe.  It connects to the phone using it's existing Bluetooth setup (I also have the bluetooth paired for AndroidAuto-Dongle.

I have the WiFi connected to AAWirelessDongle (and it has no Internet connection), but is connected.

The head end does not change  from the "Phone" icon to the "Android Auto" icon, which would happen if I used the USB cable on my phone to connect.

I just read through the information for connecting, SSH and will try to pull the logs from it!

This looks SOOO promising.. I am exited that I might be able to use Android Auto without having to connect the USB cable every time.

bradhmn, 2024-08-29:
[AAlog.txt](https://github.com/user-attachments/files/16802108/AAlog.txt)
Attached is the log i captured.   I can also add some screen shots of my android phone-- in zip file.  Pixel 3A.
[Photos-001.zip](https://github.com/user-attachments/files/16802151/Photos-001.zip)
hope its simple..   BlueTooth is connected to both the 'Santa Fe' and "AA Wireless DOngle'.  From the display on the Santa Fe it shows a phone connection, which is normal when only the Bluetooth phone is connected..  It changes to the Android Auto when it detects it..  Which isn't happening.    
Thanks