nisargjhaveri/WirelessAndroidAutoDongle #151: WiFi doesn't start on RPi Zero W

LiveSparks, 2024-08-26:
Device: Raspberry Pi Zero W
Car: Tata Nexon 2020
Firmware Version: 0.4.1
Issue:
During the connection process, Bluetooth starts and my phone is able to connect to it. After a moment, the bluetooth disconnects automatically. A Android Auto notification comes up on the Phone saying "Looking for Android Auto". This notification stays as long as the RPi has power. Besides this nothing further happens. Once in a while the head unit says "Device not Responding" but that behavior is not consistent. 
I tried to get the logs but there is no WiFi AP created by the RPi during the whole process. I checked with my phone as well as laptop.

This issue did not occur with the RPi 4. It connected flawlessly on the first try and seems to work well.

I also tried the Zero with FW v0.3.1 and observed the same behaviour. 

I'd be happy to suggest any debugging steps that you might have in mind. For now, I am leaving the Pi4 in the car. Thank you for your amazing effort.

nisargjhaveri, 2024-08-26:
Wifi not starting is strange. You can try connecting to some other power supply and see if the Wifi starts okay? If not, logs would likely help a lot.

LiveSparks, 2024-08-28:
Just now tried connecting to different power supplies including: my pc, powered usb hub, phone charger. Tried both the USB and PWR port. No change in behaviour. Bluetooth comes up but WiFi doesn't. I also tried 4 different USB cables with no change. 

I tried getting the logs via the serial interface by following this guide: [link](https://cloudchirp.medium.com/serial-to-raspberry-pi-da635122b4d0)
But I am not getting anything on serial either.
I am now going to try older versions of the firmware and see if there is a change in behaviour.

LiveSparks, 2024-08-28:
Tried different versions of the firware and even a different SD Card. No change.

nisargjhaveri, 2024-08-28:
If you can attach a screen and keyboard/mouse, that might help. Or you can use usb ethernet and connect to get the logs. Or change the location of the logs in syslogd and get logs from sd card later.

Does wifi work on the same board if you use let say Raspberry Pi OS?

LiveSparks, 2024-08-28:
Tried Raspberry Pi OS. WiFi doesn't work. I know for a fact that WiFi used to work so I am not sure what has changed. I even tried legacy versions of raspbian.

nisargjhaveri, 2024-08-28:
No problem! Thanks for testing and updating here. I'll close the issue for now.

closed 2024-08-28 by nisargjhaveri