nisargjhaveri/WirelessAndroidAutoDongle #153: MG MG5 EV - no strategy works

luisfcorreia, 2024-08-26:
Hi,

thanks for this project and all the effort involved!

I've now tried all three connect strategies and none of them work.
It seems to me that the USB thing need to be turned off completely until phone is ready.

Tested with a brand new Raspi Zero 2 W and an OnePlus 8T

I can debug this further with a laptop connected to the TTL Serial port but I'll need some help on what to look into.

Let me know, happy to help make this a better project

luisfcorreia, 2024-08-26:
Update:

while connecting this to a Linux machine I see that the USB thing isn't created with anything it can understand

Aug 26 12:43:35 lc kernel: usb 5-1.4: new full-speed USB device number 23 using xhci_hcd
Aug 26 12:43:51 lc kernel: usb 5-1.4: device descriptor read/64, error -110
Aug 26 12:44:07 lc kernel: usb 5-1.4: device descriptor read/64, error -110
Aug 26 12:44:07 lc kernel: usb 5-1.4: new full-speed USB device number 24 using xhci_hcd
Aug 26 12:44:23 lc kernel: usb 5-1.4: device descriptor read/64, error -110
Aug 26 12:44:38 lc kernel: usb 5-1.4: device descriptor read/64, error -110
Aug 26 12:44:38 lc kernel: usb 5-1-port4: attempt power cycle
Aug 26 12:44:39 lc kernel: usb 5-1.4: new full-speed USB device number 25 using xhci_hcd
Aug 26 12:44:44 lc kernel: xhci_hcd 0000:0c:00.4: Timeout while waiting for setup device command
Aug 26 12:44:50 lc kernel: xhci_hcd 0000:0c:00.4: Timeout while waiting for setup device command
Aug 26 12:44:50 lc kernel: usb 5-1.4: device not accepting address 25, error -62
Aug 26 12:44:50 lc kernel: usb 5-1.4: WARN: invalid context state for evaluate context command.
Aug 26 12:44:50 lc kernel: usb 5-1.4: new full-speed USB device number 26 using xhci_hcd
Aug 26 12:44:56 lc kernel: xhci_hcd 0000:0c:00.4: Timeout while waiting for setup device command
Aug 26 12:45:01 lc kernel: xhci_hcd 0000:0c:00.4: Timeout while waiting for setup device command
Aug 26 12:45:02 lc kernel: usb 5-1.4: device not accepting address 26, error -62
Aug 26 12:45:02 lc kernel: usb 5-1.4: WARN: invalid context state for evaluate context command.
Aug 26 12:45:02 lc kernel: usb 5-1-port4: unable to enumerate USB device

nisargjhaveri, 2024-08-26:
We don't start USB until phone is connected via TCP in phone-first and dongle mode. Is that not the case?

To try out on laptop, you can use the [Desktop Head Unit](https://developer.android.com/training/cars/testing/dhu#connection-aoap). It is a little tricky to test dongle mode with this, but the other two modes should be fairly simple to test out with this.

luisfcorreia, 2024-08-27:
no, as of today, this isn't working at all on my car

I will have to revisit this later and post back tesults

luisfcorreia, 2024-08-27:
So, here's an interesting turn of events.

I've setup Android Head Unit (AHU) with Android's SDK Manager in my desktop.

Activated the unit with "-u" for usb support and plugged the raspberry pi zero 2 w

After some back and forth with setting up phone, restarting AHU multiple times, it suddenly started working

I then went to my car and as a "miracle" it started to work!!!

I now want to know what changed to make it work in the car

nisargjhaveri, 2024-09-19:
@luisfcorreia Any insights on what changed that made it work? If not, can we close this issue if it now works for you?

luisfcorreia, 2024-09-19:
It is now less reliable then using the cable, I've stopped using it.

I'll close the issue for now, will revise it later

closed 2024-09-19 by luisfcorreia