nisargjhaveri/WirelessAndroidAutoDongle #154: Poor sound quality - RPi0W on Toyota Rav4 2020r

Ari9310, 2024-08-26:
Hi, first of all a great project, I rode for a dozen minutes so far, but I did not experience any lag, the latency was tolerable, as well as after usb cable.

The problem I have is with the noticeably weaker sound quality, the bass sounds like it's too loud and you can hear it crackle. It's like the volume on the phone is set to 120% and overdrives the range of the car radio and on the radio the quieter it is the more clearly you can hear it. I tried in the Android Auto developer settings to change the audio codec to PCM and AAC-LC but I couldn't hear any difference.

My phone is a Nothing Phone 2 on Android 14 and I listen to music through YT Music. Wired audio is as good as it gets. Is there somewhere I can change something to fix this?

And I have an additional question; by changing RPiZeroW to PRiZero2W anything will I gain? Will it at least boot faster or is it rather irrelevant?

nisargjhaveri, 2024-08-27:
RPi Zero 2 W should generally be faster to boot compared to RPi Zero W.

For the music quality, the dongle does not really change anything in the communication between the phone and the headunit. We simply proxy all communication. Lag is a common issue, likely due Wifi interference, which could be higher on 2.4 GHz networks. Other than that, I'm not sure why the audio quality would be different, unless phone is treating wireless connection differently.

Ari9310, 2024-08-27:
Very sorry for the confusion, it's all clear now. It turned out that yesterday morning, due to the broken sound in one of the podcasts, I set the mono sound on my phone and only today I was reminded of it.... Unchecking this option fixed everything.

I'll give a tip in addition, you can add somewhere that the usb otg cable differs only in bridging the GND signal to the Sense also I on my Raspberry Pi took a soldering iron and tin connected the marked pins so that I connect to the car with just one regular USB-A-MicroUSB cable without any additional adapters or OTG cables.
[image]

I saw that the topic has already been touched on but I would also like the ability to change the dpi from 160 to 140, in my opinion it would look much better with a vertical bar rather than a horizontal one as in the screengrab from the Carlinkit site.
[image]

nisargjhaveri, 2024-08-27:
> Very sorry for the confusion, it's all clear now. It turned out that yesterday morning, due to the broken sound in one of the podcasts, I set the mono sound on my phone and only today I was reminded of it.... Unchecking this option fixed everything.

Great! Thanks for confirming!

For the otg cable, I'm slightly confused. We anyway need a simple microusb to usb data cable. Why is bridging these two pins required?

For the dpi, while it might be possible to do, it requires some kind of man-in-the-middle manipulation of the Android Auto communication. We don't do that at least for now.

Ari9310, 2024-08-27:
I looked for usb otg cable diagram and found something like this:
[image]

The cable is no different from a regular cable except for this jumper and a female usb connector and not a male one. I thought that making a jumper on the connector and not in the cable should not change anything and it doesn't ;) regular cable also works

nisargjhaveri, 2024-08-28:
Thanks for explaining! If I'm not mistaken, otg cable is needed when we want the device to work in host mode. But in out case, we want the RPi to behave as usb device/gadget and not host. We should only require regular micro usb cable, and it should work well with that as well.

The common issue with micro usb cable is between the regular data cable and power-only cable, where the data lines are not connected and just the two power lines are available. That doesn't work.

nisargjhaveri, 2024-09-08:
Closing this since the original issue was resolved. Thanks for sharing the USB OTG information as well :)

closed 2024-09-08 by nisargjhaveri