nisargjhaveri/WirelessAndroidAutoDongle #155: Android Auto stops working after a phone call - 2020 Mazda CX-5

pinozero21, 2024-08-28:
Hello @nisargjhaveri, thank you for all the hard work putting this project together. Very much appreciated 👍 

I'm currently using a Raspberry Pi Zero 2 W using version 0.4.1. 

The reliability has greatly improved from version 0.4.0 in my experience with my current infotainment system. AA now connects on the first try and is stable. The only issue that I still get and I'm not able to resolve by changing settings in AA developer settings, etc is the AA disconnect from the infotainment system after making a call. 

Once I hang up, I can see the call is done via AA, then the disconnection happens, the infotainment screen shows a mobile error, and AA no longer connects until I power cycle the Raspberry Pi Zero 2 W.

Any advice on any possible troubleshooting will be greatly appreciated. Thank you and have a good day!

Mozzie-AU, 2024-08-30:
Ditto for me. 
Doesnt happen all the time, but does require a power cycle of the Zero 2 W to recover.
My setup is MG4 EV, Zero 2 W, Build version 0.41. Al version 12.5.643054

nisargjhaveri, 2024-08-30:
Any chance you can pull out the logs once it disconnects? I'd like to know what fails and kills the connection.

But yes, this is strange. Can't think of a reason why it would fail after the call is complete.

pinozero21, 2024-08-30:
@nisargjhaveri I was able to copy the following from the messages file 

[image]

nisargjhaveri, 2024-08-31:
> Jan  1 00:01:47 buildroot daemon.info hostapd: wlan0: STA f6:08:5f:6f:2a:8f IEEE 802.11: disassociated

Is this when the connection dropped? Or did you manually disconnect wifi after the connection dropped? This is the only sign, and it simply says wifi connection dropped for some reason.

What is also puzzling is that why did the recovery mechanism did not detect TCP failure and try to recover.

pinozero21, 2024-09-01:
That is also one of the questions I had, why did the pi not try to reconnect after the disconnect occurred?

I'll test again and recapture logs to see if they show anything else that can point to the root cause. It is just a weird behavior for the connection to stop after the call is done.

pinozero21, 2024-09-03:
Hi @nisargjhaveri here is the log I captured today. Same behavior, connection drops when hanging up from a call. 

[image]

nisargjhaveri, 2024-09-04:
The log seems to be partial. Does not contain the actual connection part.

pinozero21, 2024-09-06:
Hi @nisargjhaveri sorry for the confusion. Seems like the copy/paste cannot handle the entire log. Tried to use SCP to copy the messages file to my Documents folder, but no luck. 

If you could please advise on the proper procedure for copying the messages file, it will be greatly appreciated.

nisargjhaveri, 2024-09-06:
scp should've worked! :) If not, if you copy/paste the longest you can but from the end, that might also help.

Andre1902, 2024-10-27:
Same issue here with MIB3 Audi A1 and rpi 2W. After a call starts it only can stay connected for a few seconds, then it gets disconnected.

My configuration is not really standard... To shut down completely AA after switch off the car im getting power directly from 12V and not from the infotainment system that is only connected on a data cable to the dongle. Btw everything else works fine.

Doesn't AA require bluetooth for phone calls (and only for that)? My S21 FE it's not connected to the dongle after the initial connection, but it's still connected to the car's inbuilt infotainment system via bluetooth.

Thank you :)

nisargjhaveri, 2024-11-05:
> Doesn't AA require bluetooth for phone calls (and only for that)? My S21 FE it's not connected to the dongle after the initial connection, but it's still connected to the car's inbuilt infotainment system via bluetooth.

Right. The phone should be connected to the car for phone calls via bluetooth.

Not really sure on why it drops the connection after a phone call for you.

nisargjhaveri, 2024-11-05:
Does disconnection after on phone call also happens with 5GHz wifi (RPi 3A+ or RPi 4)?

It might be possible that BT and 2.4GHz Wifi might be interfering with each other causing the disconnect?

Andre1902, 2024-11-05:
> Right. The phone should be connected to the car for phone calls via bluetooth.
>
> Not really sure on why it drops the connection after a phone call for you.

I plugged in my chinese dongle. It seems that the bluetooth connection is stable with the smartphone. I guess to make phone calls possible.
There's no what to keep the bluetooth connection active by modifying some parameters, or maybe an older version of your project?
Thanks

nisargjhaveri, 2024-11-06:
Only BT connection to phone is required, connection with dongle is not used anyway in cas eof phones.

Yu can try out the dongle mode by changing the config file as well. It may have other improvements also which might help phone audio reliability.

pinozero21, 2024-12-07:
@nisargjhaveri  sorry for the delay in any reply or input. But, recently I updated my device to a Pixel 9 pro XL and changed my cable to a USB A/MicroUSB (previously had a on the go dongle and adapter to USB C)

So far the connection has been stable and AA no longer disconnects after making a call. 

If I notice any other bug, I'll post it. But, as of now, the issue on this post is no longer occurring. 

Thank you!

closed 2024-12-09 by pinozero21