nisargjhaveri/WirelessAndroidAutoDongle #158: AA Stopped Working Connection Loop

NabeelUppel, 2024-08-29:
Hi, I used this for a couple of months and it worked great but one day it stopped working in my car and I cannot get it working again.
The phone would flash the android auto screen and disappear.  

I tried:
- using the latest version and older ones. 
- switching cables (used the same one that used to work on too)
- different phones
- different connect modes
 - Dongle mode: Fails to connect to bluetooth device
 - Phone first: Flashes as mentioned above
 - USB first: Bluetooth does not show up as a device to connect to. 

Wired AA works perfectly fine. 
I cannot for the life of me figure out what broke it. 

On RPI 0 2W. 
Mazda 3 2015

Logs attached below, any help would really be appreciated. 
[logs.txt](https://github.com/user-attachments/files/16800088/logs.txt)

NabeelUppel, 2024-08-29:
Tried a different car and it worked perfectly. Not sure what's causing this in my car.

I do notice that when I initially connect it to via USB (for the first time) it says it's been disconnected, then I have to to remove the USB and reinsert it for it to be picked up. Wonder if it's related.

NabeelUppel, 2024-08-29:
Here's the logs for the other two modes
[dongleModeLogs.txt](https://github.com/user-attachments/files/16801785/dongleModeLogs.txt)
[UsbModeLogs.txt](https://github.com/user-attachments/files/16801786/UsbModeLogs.txt)

pinozero21, 2024-08-29:
@NabeelUppel which image version are you using? 

That scenario happened to me while using version 0.4.0.

Once I upgraded to v0.4.1, the connection happens on the first try and don't have to power cycle my pi02w. 

2020 Mazda CX-5

NabeelUppel, 2024-08-29:
unfortunately I'm also using v.0.4.1.

ganesh3s3, 2024-09-01:
+1 same issue. I have tried 0.4.0, 0.4.1 and even 0.3.1. Nothing works.

nisargjhaveri, 2024-09-19:
@NabeelUppel sorry for the late reply. Both your logs simply suggests to the USB connection failing with the car. The most common issue is bad cable or bad connection. Maybe you've ruled that out though. Anything changed on the headunit?

naby-dev, 2024-09-19:
I did rule out the cable, it must be related to my headunit.

I tried it in a different car and it worked fine, I just don't understand what changed because it worked perfectly for a couple of months and one day all of a sudden it stopped working.

nisargjhaveri, 2024-09-19:
Does wired android auto still work on the same car? Can the usb port be damaged to connect properly with the wire you're using with RPi? Random guesses :)

naby-dev, 2024-09-19:
Wired AA works perfectly fine.
I thought about that too. 
I'm using a RPI2 without type C and I can't seem to fine a microUSB to type C adaptor.

pinozero21, 2024-09-23:
@naby-dev I am using the following adapter and it has been working for me.

I'm using RPIZERO2W with OTG micro USB to female USB cable then the kit to adapt to USB C cable. 

HIEbee USB 3.0 A to A Male Adapter, USB Female to Female Adapter, USB Type-C to USB A Male Adapter, USB 3.0 Female to USB C Female Adapter Connector Compatible with Hard Drive, Laptop etc https://a.co/d/8Fnutqd

[image]