nisargjhaveri/WirelessAndroidAutoDongle #160: Connected with Digipad 2.0 under Honda Jazz

mohitkum05, 2024-08-30:
Hi,

I used Raspberry Pi 3 A+ over Honda Jazz Digipad 2.0 and based on steps mentioned under readme file, and was able to connect it successfully. Tried calls and it works, also tried google assistant and it worked as well.

I have below query:
1. When Wifi is on and bluetooth is off - it doesn't connect.
2. When Wifi is off and bluetooth is on - it connects and then makes wifi connection active (even though it's disabled).

Isn't it should connect to wifi post boot up and then switch on the bluetooth if it's off? So basically, if either of the connection is active it should connect and made other connection on.

For location it should be enabled in all case or else it will ask for location permission over the phone and we have to enable it but this is the case for wired connection as well.

Please let me know thoughts on this.

Thanks.

nisargjhaveri, 2024-08-30:
This is normal Android Auto behaviour that we don't control. Android Auto only gets activated on phone when particular type of bluetooth device connects. Once bluetooth is connected, it exchanges wifi connection information automatically and connects to the wifi automatically. It bluetooth is not enabled, it cannot know that the wifi is for Android auto, or the wifi password. Since the wifi often does not have internet, users are anyway not expected to connect manually to the wifi.

mohitkum05, 2024-08-30:
Okay, this is great. I wasn't aware of this behavior. Thanks a lot for this detailed clarification.  

Also, I was wondering what if head unit is having Wi-Fi support. Does it will disconnect Wi-Fi of head unit if already connected and then connect to wireless android auto Wi-Fi after connecting to Bluetooth? For Bluetooth I noticed, it disconnects head unit Bluetooth connection if already present connects to AAWireless dongle Bluetooth connection and then connects to Wi-Fi and then head unit Bluetooth connection is active again. 

I haven't tried from my end but my head unit has dongle support for Wi-Fi hotspot.

nisargjhaveri, 2024-08-30:
> For Bluetooth I noticed, it disconnects head unit Bluetooth connection if already present connects to AAWireless dongle Bluetooth connection and then connects to Wi-Fi and then head unit Bluetooth connection is active again.

This is new to me. I thought most phones can connect to multiple bluetooth at once. And it should keep the headunit bluetooth connected. Just for experimentation, you can maybe also try dongle mode and how this behaves, because it requires both bluetooth devices to be connected at once.

> Also, I was wondering what if head unit is having Wi-Fi support. Does it will disconnect Wi-Fi of head unit if already connected and then connect to wireless android auto Wi-Fi after connecting to Bluetooth?

I also haven't tested this. But I believe many recent phones do have ability to connect to more than one wifi network at once. Maybe it can keep headunit wifi connected for internet and dongle for Android Auto, not sure. For the phones that don't support multiple wifi, it'll disconnect from the headunit wifi and use mobile network for internet.

mohitkum05, 2024-08-31:
> > For Bluetooth I noticed, it disconnects head unit Bluetooth connection if already present connects to AAWireless dongle Bluetooth connection and then connects to Wi-Fi and then head unit Bluetooth connection is active again.
> 
> This is new to me. I thought most phones can connect to multiple bluetooth at once. And it should keep the headunit bluetooth connected. Just for experimentation, you can maybe also try dongle mode and how this behaves, because it requires both bluetooth devices to be connected at once.
>>> You are correct. I noticed carefully and then found that it connects to my headunit first followed by AAWireless dongle and then connects to Wifi and then bluetooth for AAWirelss gets diconnected. So I believe it's working as per standard.

> > Also, I was wondering what if head unit is having Wi-Fi support. Does it will disconnect Wi-Fi of head unit if already connected and then connect to wireless android auto Wi-Fi after connecting to Bluetooth?
> 
> I also haven't tested this. But I believe many recent phones do have ability to connect to more than one wifi network at once. Maybe it can keep headunit wifi connected for internet and dongle for Android Auto, not sure. For the phones that don't support multiple wifi, it'll disconnect from the headunit wifi and use mobile network for internet.
>>> I tested this today and found that both wifi are connected and working simultaneously. So headunit is being tethered via my phone hotspot and AAWireless Wifi is also connected as expected. Both were able to consume data / from my phone. So this too works as expected.

One last query from my end, I believe Raspberry Pi 3 A+ has capability of 5 GHz and 2.4 GHz connections were other Pi boards zero W or zero 2W uses 2.4 GHz based on their capability. Also, I think I was connected 5 GHz connection based on my checks performed when I was connected with AA Wireless. Can you please confirm if this will always be the case that it will use the highest possible channel (2.4 GHz / 5 GHz) for WiFi connection based on boards capability?

mohitkum05, 2024-08-31:
> > Can you please confirm if this will always be the case that it will use the highest possible channel (2.4 GHz / 5 GHz) for WiFi connection based on boards capability?
> 
> Not automatically. But currently the RPi 3A+ and RPi 4 images are configured to use the 5GHz channel, and the other two boards are configured to use the 2.4GHz channel by default. If we add a new board, we need to configure the image correctly to use the appropriate channel as available.

This is great. Thanks a lot again for clarifying. Seems investing in good board works as 5 GHz is fast and noise free. I see no lag in music play, phone communication and navigation handling. All works seamlessly. Thanks a lot for sharing your good work and keep it up. I started using this more frequently than wired connection due to effortless connection. I will let you know if I have further feedback / related queries / issues. 

Once again, thank you and closing this thread.

closed 2024-08-31 by mohitkum05