nisargjhaveri/WirelessAndroidAutoDongle #163: Auto shutdown with car

sushilmudgil, 2024-08-31:
HI Nisarg, 

wanted to thank you for this awesome project.  

I was able to run it on a combination of **Raspberry Pi Zero 2 W** and Suzuki XL6 Zeta 6 AT. It pairs almost instantly without any lag and delay.  You can add **_XL6 ZETA HU_** to the list of compatible cars .

My Only question is - 

A) The moment i switch off the Car ignition, it should switch off the PI as well but i see PI keeps running thus draining the battery. Is this normal behaviour and if this impacts the longevity of device in any way.  is there any way that we can write in code that if there is no inut for 2 min ( example) it should shut it self down.

regards
Sushil

nisargjhaveri, 2024-08-31:
The assumption is that most cars will kill power on USB once ignition is off. But I know this is not true for all.

We can turn off the device when inactive. But if the power is always on, we might not have a nice way to restart the device automatically once the car starts again as it'd need a power cycle to start again. Any ideas on how and when we can restart the device again?

sushilmudgil, 2024-09-03:
> > The assumption is that most cars will kill power on USB once ignition is off. But I know this is not true for all.
> Any ideas on how and when we can restart the device again?

 I would expect a code to integrate it with car ignition which i think might not be possible as we have no access to Car ECU . 

so way around would be, for it to detect HU power state and proceed accordingly. So If HU is powered on.. it powered itself, If Off it switches itself Off.

nisargjhaveri, 2024-09-03:
> so way around would be, for it to detect HU power state and proceed accordingly. So If HU is powered on.. it powered itself, If Off it switches itself Off.

I don't think we generally have an easy way to detect if HU is on/off, if the USB power is always on, especially if we shutdown the dongle in the first place.

rmccaskill, 2024-09-04:
My personal experience with Kia Sorento. The USB power shuts off a minute or two after you lock the doors. When you unlock the doors, the USB is powered again.

wsadzc, 2024-09-17:
There's a crude solution to this problem:
Purchase a 12V relay and a cigarette lighter plug.
Disconnect the Pi's VCC 5V and connect it to the relay's control.
When the car starts, the cigarette lighter activates, the relay engages and conducts electricity.
The 5V VCC powers up, and the Pi starts.
This can achieve the functionality you want, but it's a rough solution.
Alternatively, you could purchase an ELM327 OBDII communicator to read the CAN bus signal for startup.
This is much more difficult and would be a different functionality altogether.
The above conversation was translated with the assistance of Claude.ai. I hope I've expressed my meaning clearly.
Thank you.

utsavsinghal, 2024-09-21:
My experience with Tata Altroz is the car shuts off USB power ~5mins after the doors are locked. 

My only concern is that we are not doing shutdown safely as the power is abrubtly switched off after locking the car. This may impact the raspberry pi board or the memory card in the long run.

nistvan86, 2024-09-27:
In case the file system on the SD card is read only, and the full mini Linux environment runs entirely from the Pi's memory, sudden cutting the power is probably not an issue. The Raspberry Pi OS has a read only fs implementation just for this.

Does this project writes to the SD card anything during operation? I can imagine, there's some data related to the initial device pairing. But after that I guess it could run readonly. Maybe an initial setup could write that data to the SD card then switch to fully read only mode.

nistvan86, 2024-09-27:
I think I've found the answer. In the buildroot config the root file system is [not configured for read-write remount](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/main/aa_wireless_dongle/configs/raspberrypizero2w_defconfig#L75C1-L75C37), and also [the copied fstab file also specifies everything as ro](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/main/aa_wireless_dongle/board/raspberrypi/rootfs_overlay/etc/fstab#L2). So this whole Linux environment and the proxy runs from memory. I think there should be no problem with cutting the power then.

rawdr, 2024-10-04:
Could the Pi scan for the cars Bluetooth Mac, and choose to activate or sleep based on that?

abhi18588, 2024-10-11:
> HI Nisarg,
>
> wanted to thank you for this awesome project.
>
> I was able to run it on a combination of **Raspberry Pi Zero 2 W** and Suzuki XL6 Zeta 6 AT. It pairs almost instantly without any lag and [...]

Hello @sushilmudgil , Suzuki cars shutoff the poower to the Android Auto in approximately 30 to 60 seconds after the ignition is off and the doors are locked from outside. I have a 2019 Baleno and the Pi BOard shuts off in approximately 30 seconds after doors are locked. Kindly check and it should be the same in XL6 as well.

puregreen59, 2024-11-14:
I've got a Renault Zoe (EV) that's parked on my drive.

the USB ports seem to be either always powered or powered when charging.

this means that my phone connects to it while I'm near the front of my house and even upstairs in the bedroom and I have to either remember to unplug it when I leave the car or turn off WiFi and Bluetooth on my phone.

The super hacky but unfriendly way to the pi that I'm planning on trying Is a usb cable with a switch inline.

 I'll still need to remember to switch it to off before leaving the car but this is at least slightly more friendly than unplugging and plugging a usb cable in each time

> Could the Pi scan for the cars Bluetooth Mac, and choose to activate or sleep based on that?

This is a great idea as my cars Bluetooth turns of when I open the door and would totally solve me issue!

It could scan for my Renault Bluetooth and if its found can resume normal operation, if it doesn't it goes into standby until it sees Renault again?

sivachandran, 2024-11-15:
@puregreen59 I faced the same problem and solved with [this](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/211) setup.

sushilmudgil, 2024-11-15:
> Hello @sushilmudgil , Suzuki cars shutoff the poower to the Android Auto in approximately 30 to 60 seconds after the ignition is off and the doors are locked from outside. I have a 2019 Baleno and [...]

@abhi18588 you are right . I realised later on. Just forget to update here Between i did few really long drives .. 10 hrs at stretch and it did not even crashed or rebooted once with maps & amazon music on all the time.

closed 2024-11-15 by sushilmudgil