nisargjhaveri/WirelessAndroidAutoDongle #164: Connection keeps on breaking every now and then. Tried different cables and SD cards same problem.

sumitk31, 2024-09-01:


ramon2k10, 2024-09-01:
The same for me in a Class a Mercedes

rameshandroid, 2024-09-01:
i am also facing the same issue. i tested it on  raspi4 model-B . we tried multiple versions. its keep on showing looking for android  auto. any help would be great.i can share the logs if required

nisargjhaveri, 2024-09-02:
Couple of potential issues:
- Inadequate power supply or bad usb cable.
- Wifi interference, bad enough to kill the connection.

Logs might help better to figure out the issue.

nisargjhaveri, 2024-09-02:
> i am also facing the same issue. i tested it on raspi4 model-B . we tried multiple versions. its keep on showing looking for android auto. any help would be great.i can share the logs if required

This seems different than the original issue. See the troubleshooting in README. If that does not help, feel free to create another issue with logs.