nisargjhaveri/WirelessAndroidAutoDongle #165: Support for Banna PI Zero

tecnomanu, 2024-09-01:
Hi, i'm trying to insert the image for Raspberry Pi Zero in my Banana Pi Zero (same model, different brand) but is not working, can anyone test for this model already?

Thanks!

nisargjhaveri, 2024-09-02:
While the name and form factor are similar, the two boards are fairly different. The images pre-built for RPi will not work on this board.

https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/162#issuecomment-2322884930 contains some information on how to go about supporting a new board. Hope this helps.

tecnomanu, 2024-09-07:
Hi, i'm trying with an Banana Pi M2 Zero, this board have bluetooth and WIFI, because already i use it for another projects. I try to recreate the boards with the default options on buildroot but i understand the patchs in the repo are most important to convert in a android device for the car, but with many intents, the board not generete anything, i coulnd't try check with HDMI if something is happend. 

My config is this:
```
# Configuración de Arquitectura
BR2_arm=y
BR2_cortex_a7=y
# BR2_ARM_EABIHF=y
BR2_ARM_FPU_NEON_VFPV4=y
BR2_PACKAGE_HOST_LINUX_HEADERS_CUSTOM_5_9=y

# Mensaje de Bienvenida
BR2_TARGET_GENERIC_ISSUE="Welcome to Buildroot for the Bananapi M2 Zero"
BR2_ROOTFS_POST_IMAGE_SCRIPT="support/scripts/genimage.sh"
BR2_ROOTFS_POST_SCRIPT_ARGS="-c board/bananapi/bananapi-m2-zero/genimage.cfg"

# Configuración del Kernel
BR2_LINUX_KERNEL=y
BR2_LINUX_KERNEL_CUSTOM_VERSION=y
BR2_LINUX_KERNEL_CUSTOM_VERSION_VALUE="5.9.11"
BR2_LINUX_KERNEL_DEFCONFIG="sunxi"
BR2_LINUX_KERNEL_CONFIG_FRAGMENT_FILES="$(BR2_EXTERNAL_AA_WIRELESS_DONGLE_PATH)/board/common/kernel.config"
BR2_LINUX_KERNEL_LZ4=y
BR2_LINUX_KERNEL_NEEDS_HOST_OPENSSL=y
BR2_TARGET_ROOTFS_EXT2=y
BR2_TARGET_ROOTFS_EXT2_4=y
BR2_TARGET_ROOTFS_EXT2_SIZE="120M"

BR2_LINUX_KERNEL_DTS_SUPPORT=y
BR2_LINUX_KERNEL_INTREE_DTS_NAME="sun8i-h2-plus-bananapi-m2-zero"

# Rootfs overlay
BR2_ROOTFS_OVERLAY="$(BR2_EXTERNAL_AA_WIRELESS_DONGLE_PATH)/board/common/rootfs_overlay $(BR2_EXTERNAL_AA_WIRELESS_DONGLE_PATH)/board/bananapizero/rootfs_overlay"
BR2_GLOBAL_PATCH_DIR="$(BR2_EXTERNAL_AA_WIRELESS_DONGLE_PATH)/patches"

# Soporte para módulos del kernel comprimidos
BR2_PACKAGE_BUSYBOX_SHOW_OTHERS=y
BR2_PACKAGE_KMOD=y
BR2_PACKAGE_KMOD_TOOLS=y
BR2_PACKAGE_XZ=y
BR2_PACKAGE_HOST_KMOD_XZ=y

# Configuración de U-Boot (Bootloader)
BR2_TARGET_UBOOT=y
BR2_TARGET_UBOOT_BOARDNAME="bananapi_m2_zero"
BR2_TARGET_UBOOT_CUSTOM_VERSION=y
BR2_TARGET_UBOOT_CUSTOM_VERSION_VALUE="2020.10"
BR2_TARGET_UBOOT_NEEDS_DTC=y
BR2_TARGET_UBOOT_NEEDS_PYTHON3=y
BR2_TARGET_UBOOT_NEEDS_PYLIBFDT=y
BR2_TARGET_UBOOT_FORMAT_CUSTOM=y
BR2_TARGET_UBOOT_FORMAT_CUSTOM_NAME="u-boot-sunxi-with-spl.bin"

# Herramientas adicionales para U-Boot
BR2_PACKAGE_HOST_UBOOT_TOOLS=y
BR2_PACKAGE_HOST_UBOOT_TOOLS_BOOT_SCRIPT=y
BR2_PACKAGE_HOST_UBOOT_TOOLS_BOOT_SCRIPT_SOURCE="board/bananapi/bananapi-m2-zero/boot.cmd"

# # Firmware específico para Banana Pi
BR2_PACKAGE_LINUX_FIRMWARE=y
BR2_PACKAGE_LINUX_FIRMWARE_REALTEK=y

# # Controladores WiFi y Bluetooth
# BR2_PACKAGE_RTL8189ES=y
# BR2_PACKAGE_FIRMWARE_REALTEK_RTL8189FTV=y

# Herramientas requeridas para crear la imagen SD
BR2_PACKAGE_HOST_DOSFSTOOLS=y
BR2_PACKAGE_HOST_GENIMAGE=y
BR2_PACKAGE_HOST_MTOOLS=y

# Paquetes adicionales
BR2_TOOLCHAIN_BUILDROOT_CXX=y
BR2_PACKAGE_RNG_TOOLS=y
BR2_PACKAGE_HAVEGED=y
BR2_PACKAGE_BLUEZ_TOOLS=y
BR2_PACKAGE_BLUEZ5_UTILS=y
BR2_PACKAGE_BLUEZ5_UTILS_CLIENT=y
BR2_PACKAGE_BLUEZ5_UTILS_TOOLS=y
BR2_PACKAGE_DNSMASQ=y
BR2_PACKAGE_DROPBEAR=y
BR2_PACKAGE_HOSTAPD=y
BR2_PACKAGE_WIRELESS_REGDB=y
BR2_PACKAGE_IW=y
BR2_PACKAGE_WPA_SUPPLICANT=y
BR2_PACKAGE_UTIL_LINUX=y
BR2_PACKAGE_UTIL_LINUX_RFKILL=y
BR2_PACKAGE_UMTPRD=y
BR2_PACKAGE_AAWG=y

# Otros opciones
BR2_TARGET_GENERIC_ROOT_PASSWD="bananapi"
```

The file is called bananapizero_defconfig and this the config inside the docker-compose.yml

`  
bananapi:
    <<: *generate_image
    environment:
     - BOARD_NAME=bananapizero
`

Can you see some error here maybe?

Thanks!

nisargjhaveri, 2024-09-07:
Does the build succeed with this config?

I'd suggest you first get a working build using either HDMI or console setup. It will be easier to debug. Once you have the build booting properly, you cam move on to making sure the BT/Wifi and then the aawg daemon work as expected.

tecnomanu, 2024-09-20:
Hi

> Does the build succeed with this config?
> 
> I'd suggest you first get a working build using either HDMI or console setup. It will be easier to debug. Once you have the build booting properly, you cam move on to making sure the BT/Wifi and then the aawg daemon work as expected.

Hi @nisargjhaveri, the build is success but is not working and not show anything in HDMI. I will try test and reply you again.

Kelsch, 2025-08-29:
Were you able to get this working? I have a banana pi zero that I don't really know what else to use it for lol.