nisargjhaveri/WirelessAndroidAutoDongle #168: Switch to different device

samirkape, 2024-09-04:
Hi Nisarg,

Thanks for building this awesome thing. Whole setup took hardly 10 mins and it is working perfectly!

Just wanted to check if it is possible to switch to newer device who request for connection.

If yes, I would be happy to contribute

bhushankorpe, 2024-09-04:
Which car did you try this for?

samirkape, 2024-09-04:
Hyundai elite i20 asta

bhushankorpe, 2024-09-04:
Okay. Does your multimedia UI have a dedicated android auto tab or button? Asking this with reference to a problem I am facing.

samirkape, 2024-09-04:
Yes.. my car supports Android auto in wired mode

samirkape, 2024-09-04:
Thanks for sharing.

> the only time I've had an issue with these two MCUs is when using a low quality SD card, and even that did work

Is it really the case @nisargjhaveri ? Any recommendations about SD card?

nisargjhaveri, 2024-09-04:
> Is it really the case @nisargjhaveri ? Any recommendations about SD card?

A good quality sd card can be both faster and more reliable (low chances of corruption). Though, I don't have enough research to really recommend which sd card you should use

> Just wanted to check if it is possible to switch to newer device who request for connection.

Yeah, this would be good to have. Currently, if multiple devices are paired, in phone-first mode, whichever device connects first wins. The dongle tries to connect in unspecified order as well.

In dongle mode though, the phone also needs to be connected to the headunit, and we don't have much choice on which phone to connect if multiple are paired. Haven't been able to test how it handles disconnect from the headunit and new phone is now connected.

For when the connection is already present and a new phone connects, we should be able to disconnect and start a new connection. We need to see what happens if multiple phones connect in the beginning.

Also, in phone-first mode, we disable bluetooth as soon as a phone is connected. We'd need to change that. Keeping the bluetooth always on has one potential downside of increased interference with wifi in the 2.4GHz band and also I believe the chip shares some resources. We need to experiment a bit.

samirkape, 2024-09-05:
> > > Is it really the case @nisargjhaveri ? Any recommendations about SD card?
> > 
> > A good quality sd card can be both faster and more reliable (low chances of corruption). Though, I don't have enough research to really recommend which sd card you should use
> > 
> 
> I would suggest everyone should be using a class 10 rated micro SD card if they want speed, reliability, and quality wireless AA.

But once we load OS in memory, do you think the SD card performance really matters?

closed 2024-09-15 by samirkape