nisargjhaveri/WirelessAndroidAutoDongle #169: Android auto doesn't come up in Jeep Compass 2019

bhushankorpe, 2024-09-04:
Hi,

I am using the following setup:
1. Raspberry Pi Zero W
2. 16 GB card
3. Jeep Compass Limited Plus (2019)

The vehicle multimedia supports wired android auto. But there is no fixed tab/button for Android Auto in the media unit. It appears only when an android phone is connected via USB. Is there anything we can do to bring up Android Auto in the media unit? In case this helps, I noticed that the dongle only asked for phone call permissions and nothing else. Please let me know if this can be dealt with. Thanks

vishnu3095, 2024-09-04:
I just installed on a jeep compass with a pi zero 2. had to plug both micro usb to get it working. i think the power through one port in jeep is not enough

bhushankorpe, 2024-09-04:
@vishnu3095 Thanks a lot! silly me. That should have been the first thing to look for. This resolved the issue. Thanks @nisargjhaveri ! great work.

bhushankorpe, 2024-09-04:
This thread can be closed

closed 2024-09-04 by bhushankorpe