nisargjhaveri/WirelessAndroidAutoDongle #170: Works on 208 ii but car keep spamming BT request eventually leading to a disconnect to AADongle

LorisTecnology, 2024-09-04:
So i've tested this with a Rpi 4 and works flawlessly however after a while the car start spamming for BT connection and my device disconnect from AADongle

Redmi note 10 pro with PixelOS

nisargjhaveri, 2024-09-05:
> So i've tested this with a Rpi 4 and works flawlessly however after a while the car start spamming for BT connection and my device disconnect from AADongle

Can you elaborate more on this? The car BT should already be connected when Android Auto is connected. Is that not the case?

LorisTecnology, 2024-09-05:
when i connect to the AADongle the device then connects to Peugout BT instead

nisargjhaveri, 2024-09-19:
When the Android Auto connects, it is normal behaviour that the "AA Wireless Dongle" bluetooth will disconnect and the Car's bluetooth will connect automatically.

LorisTecnology, 2024-09-19:
Using it for week now and everything works flawlessly. Thanks a lot

nisargjhaveri, 2024-09-19:
Glad that it works well for you now. I'll close the issue for now.

closed 2024-09-19 by nisargjhaveri