nisargjhaveri/WirelessAndroidAutoDongle #171: Searching for android auto only

deepakhadiya, 2024-09-05:
Car - Hyundai i10 nios Sportz 2022 model (Having wired android auto)
mobile - Oneplus6T
Board - Raspberry Pi Zero 2W (From Silverlineelectronics)

I have followed all the steps and faced multiple issues to connect with bluetooth initially.
But when i use micro usb power port of board then i am able to connect Bluetooth and wifi. also able to see androidauto connected and searching for android auto as this will not work as we have used power port we need to use usb port of board (second).

When i use the usb port of board (second) it will connect the bloetooth but wifi keep on assigning ip address never able to see the andoid auto connected.

Which means with usb port it is not able to get connected. i have also followed all steps multiple times. restarted car many times.
Also tried with all 3 version of raspberrypizero2w-sdcard.img.xz
v0.4.1
v0.4.0
v0.3.1

Let me know what should be done from my side.

sasha7460, 2024-09-05:
i am facing similar problem with my connection as well.
Car - TATA  Altroz iturbo 2021 model (Having wired android auto)
mobile - SAMSUNG S23 Ultra
Board - Raspberry Pi Zero 2W (From Silverlineelectronics)

i have successfully flashed the R Pi zero w with the image using the balenaEtcher.
i connected my phone to the car and started the Android Auto in the phone.
now i connected the R Pi to the car, with the microusb connected to the data port of R pi.
i am able to connect my phone to both the bluetooth and wireless hotspot from R pi, though the bluetooth connection is erratic and gets disconnected most of the times.

This initiates Android Auto from phone, but i see Device error message from my car's IVI.

Any suggestions on how to resolve?

nisargjhaveri, 2024-09-05:
> When i use the usb port of board (second) it will connect the bloetooth but wifi keep on assigning ip address never able to see the andoid auto connected.

Some are having an issue where the connecting the USB port alone is not able to provide enough power to the board and it require connecting another cable to power usb port. It generally results in reboot, but some reports suggests weird behaviour in BT/Wifi connection as well. I don't personally know how safe is it to connect both the cables on the RPi, but seems to work based on some of the other issues.

sbala05, 2024-09-06:
https://forums.raspberrypi.com/viewtopic.php?t=223891

Raspberry Pi forums are saying do not connect both PWR and USB as the power lines are directly connected. 

I have the same problem in one of my cars. I see that the Pi zero 2W is rebooting (flickering green light) and the car never connect. In another car it works with no issues

MridulChhajer, 2024-09-06:
@deepakhadiya does the display on your i10 shows android auto ? Mine connects to the wifi and bluetooth but does not display the android auto . Tried multiple cables but no luck .
[image]

deepakhadiya, 2024-09-06:
> @deepakhadiya does the display on your i10 shows android auto ? Mine connects to the wifi and bluetooth but does not display the android auto . Tried multiple cables but no luck . ![IMG_20240904_223035709_Original](https://private-user-images.githubusercontent.com/180642689/365109521-9997c5ab-b75b-41fe-82e5-081fd069b792.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU2MTYwMTEsIm5iZiI6MTcyNTYxNTcxMSwicGF0aCI6Ii8xODA2NDI2ODkvMzY1MTA5NTIxLTk5OTdjNWFiLWI3NWItNDFmZS04MmU1LTA4MWZkMDY5Yjc5Mi5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MDZUMDk0MTUxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTRmMThhNjE4YTQyZTUxMDFhMjczODA3M2YxNTZlMDE1M2JjYzZjOTFmMWNjNjI4NTI4YTU3NjYwYzMyZGZhYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.51WdhfxKA8y11YiCCnpGaOAGHK_a-7YCxcLwQJElMbc)

There are two scenarios i have tried but android auto doesn't appear on car head unit.
1) whith only usb port android auto keeps on searching.
2) with power and usb both port. i am able to see android auto connected in mobile but in car head unit no android auto shows up.

So actually same issue i am facing as per your screenshot.

nisargjhaveri, 2024-09-06:
Please make sure you're using good quality data cable with the USB port (not PWR port). If the issue still persists, try to get the logs? It might help.

deepakhadiya, 2024-09-06:
Hi @nisargjhaveri 

I think it shouldn't be data cable issue because when i connected the board with laptop it is able to recognised as android auto. I have used two different data cable of ambrane company. I still try tomorrow with other cable as you suggested. 

Can you please share how can i take logs, Can you please share complete DIY which will help community?

I have also checked in readme file with troubleshooting but i couldn't understand how to do it. i am new to this things.

Thank you for your response and support. 

I hope you will find root cause of this issue.

deepakhadiya, 2024-09-07:
Today I have tried with boat cable but no luck.
I have also tried boat and ambrane cable with micro USB supported phone and that phone is able to connect with Android auto so I don't think it was cables issue.
[image]
[image]
[image]
[image]

sasha7460, 2024-09-10:
> i am facing similar problem with my connection as well but i am using **zero W** Car - TATA Altroz iturbo 2021 model (Having wired android auto) mobile - SAMSUNG S23 Ultra Board - Raspberry Pi Zero W (From Silverlineelectronics)
> 
> i have successfully flashed the R Pi zero w with the image using balenaEtcher. i connected my phone to the car and started Android Auto in my phone. now i connected the R Pi to the car, with data cable connected to the data port of R pi. i am able to connect my phone to both the bluetooth and wireless hotspot of R pi, though the Bluetooth connection is erratic and gets disconnected most of the times.
> 
> This initiates Android Auto from phone, but i see Device error message from my car's IVI.
> 
> Any suggestions on how to resolve?
> 
> I tried using an additional cable for power as discussed in some of the solutions, but still same.
> 
> ![20240905_171826.jpg](https://private-user-images.githubusercontent.com/52835523/364765703-fa4a860e-1f07-4400-be80-bc325e4a9c9e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU5NjA5MjYsIm5iZiI6MTcyNTk2MDYyNiwicGF0aCI6Ii81MjgzNTUyMy8zNjQ3NjU3MDMtZmE0YTg2MGUtMWYwNy00NDAwLWJlODAtYmMzMjVlNGE5YzllLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTEwVDA5MzAyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRkMTA0OWVjZjFjYTQ3YWVmYmJmMzU3ZDUyNzVlZmQzMGQyYTBjNjIxYjVhNzU2ZGFkMzAzY2YzOGI1MzgwOGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.1RJC1BwxorbWTEE09w_5p2dNWumHy1fr2fD9J6pUGwM)
> 
> ![Screenshot_20240905_171917_Settings.jpg](https://private-user-images.githubusercontent.com/52835523/364765702-6a6115a7-1a15-4a6b-9062-12bbd56b2b85.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU5NjA5MjYsIm5iZiI6MTcyNTk2MDYyNiwicGF0aCI6Ii81MjgzNTUyMy8zNjQ3NjU3MDItNmE2MTE1YTctMWExNS00YTZiLTkwNjItMTJiYmQ1NmIyYjg1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTEwVDA5MzAyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBjOTExNTE3MmJjNzg5NThjM2JhY2IxN2FmM2ViNzUzYmY4ZmYwMDQzYzFmYTM2ZWJiZDMzMWU3ZDk3YmIxOTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xKrmq_QTN748tYUpwN9nsqAGkFzFmPcag14g-9Dybbo)

i was able to get the log for this from the pi.
[log.pdf](https://github.com/user-attachments/files/16942670/log.pdf)
would this point to insufficient power supply for the unstable connection?

sasha7460, 2024-09-10:
[update]: i connected the board to a friend's innova hycross and it works flawlessly
so i think the issue should be with the TATA Altroz's head unit. 
i will start a separate thread for this issue.

MridulChhajer, 2024-10-06:
Any solution 

> Today I have tried with boat cable but no luck. I have also tried boat and ambrane cable with micro USB supported phone and that phone is able to connect with Android auto so I don't think it was cables issue. ![IMG_20240907_184008](https://private-user-images.githubusercontent.com/5277599/365380832-a6baa315-d656-4c16-aaed-e6bb4765a3e1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgyMzc4MzMsIm5iZiI6MTcyODIzNzUzMywicGF0aCI6Ii81Mjc3NTk5LzM2NTM4MDgzMi1hNmJhYTMxNS1kNjU2LTRjMTYtYWFlZC1lNmJiNDc2NWEzZTEuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMDZUMTc1ODUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Yzg1OTllMzRiYjI1Y2FiODdjMjBjMDViZGU0NjM5ZDhmZWE1MmIwZGQwNzMxNzY2MTU4ZjgwNmNmYjhiMGQyNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.wrPA_Knj0PEieZVVXHmYBntNUyw3pAexjJ0GIjUx_WE) ![Screenshot_20240907-183946](https://private-user-images.githubusercontent.com/5277599/365380890-a6d04e5c-4692-443d-a858-30863643e8a0.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgyMzc4MzMsIm5iZiI6MTcyODIzNzUzMywicGF0aCI6Ii81Mjc3NTk5LzM2NTM4MDg5MC1hNmQwNGU1Yy00NjkyLTQ0M2QtYTg1OC0zMDg2MzY0M2U4YTAuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMDZUMTc1ODUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjRmYjkzOGI0ODhlNGM5NTI3MTI3YzY0NTYyNzhjODE3ODU3YWJlZTMzNTRiYjYxNGNhNzhjMWE1NzZhYzBjZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.uqt_s8c0vgsB0c5bVMYw9Fv0kQD-vfUAOj6GqoXFNuI) ![Screenshot_20240907-183934](https://private-user-images.githubusercontent.com/5277599/365380901-8c8012dd-7b33-4d16-80d8-813060ff4e4b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgyMzc4MzMsIm5iZiI6MTcyODIzNzUzMywicGF0aCI6Ii81Mjc3NTk5LzM2NTM4MDkwMS04YzgwMTJkZC03YjMzLTRkMTYtODBkOC04MTMwNjBmZjRlNGIuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMDZUMTc1ODUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2I1MjE1YTkyNDExZDgwYTFmMTc5MjIxYjgwYzNmNTJlNzU5ZWZjMzZmYzBiM2M1ODA5M2ZiY2IzMmQ3Zjc1YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.VqoE24vN5ywO3EhqwbM6mK103oPooBEFY6cSQrxThKo) ![Screenshot_20240907-183924](https://private-user-images.githubusercontent.com/5277599/365380906-ee3404d0-bf50-4b13-a3d4-debd8bffaeba.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgyMzc4MzMsIm5iZiI6MTcyODIzNzUzMywicGF0aCI6Ii81Mjc3NTk5LzM2NTM4MDkwNi1lZTM0MDRkMC1iZjUwLTRiMTMtYTNkNC1kZWJkOGJmZmFlYmEuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMDZUMTc1ODUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZmNkZGEwZTg2MWU5ZjZhMGVhYTc2YzQzMThiYmM5NWE5MTZmYTQ3YzllNzk4NzRlNzIzNzM4ZGJiZTU2NGRlNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.JNp-Y1knFzGKgRwezJPQ7kQ-aoQZmINYz7XDFOH4Xpk)

Did you find any solution to the problem ? Multiple good quality cables are also not working for me

Subramanian-SS, 2024-10-14:
I have the same issue in my setup. The pi boots up and on connecting the USB to USB/OTG port. On pairing the device to the mobile, the connection succeeds and I see that the phones WiFi gets connected to the dongle. The Android Auto keeps trying to connect, and there are many "Device No Response" errors popping up on the car's screen.

Car: Nexon 2021 XZ+S - AAWired Enabled.
Raspberry Pi Zero 2W running v0.4.1. in a 1 GB SD Card.
Mobile connected - Moto G34 5G - Android 14.
The issue is seen with or without additional power supplied via the PWR in the Pi02W.

The files below are the syslogs from my latest attempt this evening.
[AAWirelessSyslogMessages.log](https://github.com/user-attachments/files/17367483/AAWirelessSyslogMessages.log)
[AAWirelessSyslogMessages2.log](https://github.com/user-attachments/files/17367488/AAWirelessSyslogMessages2.log)

I can confirm that the board is not rebooting as the ssh connection to the PI was not reset and I had `tail -f` on the syslog file.
Any pointers to resolve this?

Subramanian-SS, 2024-10-16:
A quick update. It works for me with a different cable.
There was an old Bose cable lying around which was shorter (1 foot). I tried with OnePlus 10R and Moto G34 5G. Both work.
Tried Voice Call, WhatsApp call, VLC, Waze, Maps and Google Assistant. All work.

Voice call when disconnected caused Android Auto to get disconnected as well. Do not have logs yet. Will try again and see if that repeats.