nisargjhaveri/WirelessAndroidAutoDongle #172: Device Not Supported error pop up 2019 Ram 1500

divyangmd, 2024-09-05:
My 2019 Ram 1500 supports wired Android Auto.  After connecting my rp zer02w with 8gb card the phone serially connects and disconnects from the AAWirelessDongle wifi and displaces Android Auto connected on the phone while the wifi is connected.
'Looking for Android Auto' message constantly shows regardless of the wifi connection.
The Ram screen will display 'Device not Supported' message.
I tried powering the rp zero2w with both otg and power ports connected to usb and with otg alone plugged in alone. The problem above still persists and the truck never shows the Android Auto icon for selection.

Car:  2019 Ram 1500
Device: Raspberry Pi Zero 2 W
Prebuilt SD Card version: 0.4.1 (also tried 0.4.0)

Any help would be appreciated.  I attached a .txt file with a copy of what was shown on the /var/log/messages file.  I couldn't sftp into the rp02w to copy the file so I copy and pasted the messages file.
[messages.txt](https://github.com/user-attachments/files/16897529/messages.txt)

sasha7460, 2024-09-05:
i think the issue is same here as well, though i am not sure where to locate the log file in mine.
[Device no response](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/171#issuecomment-2331113371)

nisargjhaveri, 2024-09-05:
Looks like car sees the USB device, but never sends the accessory start request as expected. Maybe the car is looking for some specific USB configuration? We do present the device as MTP, but it does not have any operations enabled. Maybe someone with the error can try changing some parameters there and see if it helps?

divyangmd, 2024-09-06:
Thank you for your help on this!  
Ok, I tried something new.  I turned on the rp0w2 with OTG plugged in alone.  It went through multiple cycles of Device not Supported and seemed like rp0w2 reboots.  So I then connected the power port to another USB in the truck. That did nothing for a while. So I then unplugged the OTG port and plugged it back in.  All of a sudden wireless Android Auto came on the truck!  YEAHHHH!  After it connects I can unplug the power port and the Android Auto stays connected. 
I am attaching the log - messages2

Then I started the whole process again and started with both the power and OTG plugged into the truck.  After a while nothing was happening so I unplugged the OTG port only and plugged it back in and Android Auto started! YEAH!!  I then can unplug the power port and Android Auto stays connected.
I am attaching the log - messages 3

If you have time could you look at them and see why I have to unplug and replug in the OTG for it to initialize Android Auto.
I would like for this to work before I go and change the fuse set up in the truck because right now the USB slots are always hot and I probably dont want that since it stays on for over an hour and when I am in the house my phone stays connected to the rp0w2 because it stays on, lol. 

[messages2.txt](https://github.com/user-attachments/files/16909122/messages2.txt) (Initially started with only OTG plugged in)
[messages3.txt](https://github.com/user-attachments/files/16909123/messages3.txt) (Initially stated with both power and OTG plugged in)

divyangmd, 2024-09-06:
messages2 - (Initially started with only OTG plugged in)

                                                                                  

Jan  1 00:00:03 buildroot syslog.info syslogd started: BusyBox v1.36.1

Jan  1 00:00:03 buildroot user.info console: Starting syslogd: OK

Jan  1 00:00:03 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-08-03 03:46:23 UTC)

Jan  1 00:00:03 buildroot user.info console: Starting klogd: OK

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0

Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61-v7 (runner@fv-az1424-844) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils)>

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: div instructions available: patching division code

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero 2 W Rev 1.0

Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] random: crng init done

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writealloc

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Zone ranges:

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x0000000019bfffff]

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   Normal   empty

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Movable zone start for each node

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Early memory node ranges

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] percpu: Embedded 12 pages/cpu s18004 r8192 d22956 u49152

Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s18004 r8192 d22956 u49152 alloc=12*4096

Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545

Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_f>

Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Unknown kernel command line parameters "modules-load=dwc2,g_ether", will be passed to user space.

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory: 337224K/421888K available (9216K kernel code, 718K rwdata, 2316K rodata, 1024K init, 520K bss, 19128K reserved, 65536K cma-reserved)

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1

Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.

[... 683 lines cut ...]
Jan  1 00:07:09 buildroot daemon.info hostapd: wlan0: STA f2:2e:1d:04:ae:c9 IEEE 802.11: disassociated

Jan  1 00:07:09 buildroot daemon.info hostapd: wlan0: STA f2:2e:1d:04:ae:c9 IEEE 802.11: disassociated

Jan  1 00:07:39 buildroot daemon.info hostapd: wlan0: STA f2:2e:1d:04:ae:c9 IEEE 802.11: disassociated

Jan  1 00:07:39 buildroot daemon.info hostapd: wlan0: STA f2:2e:1d:04:ae:c9 IEEE 802.11: disassociated

Jan  1 00:08:09 buildroot daemon.info hostapd: wlan0: STA f2:2e:1d:04:ae:c9 IEEE 802.11: disassociated

Jan  1 00:08:39 buildroot daemon.info hostapd: wlan0: STA f2:2e:1d:04:ae:c9 IEEE 802.11: disassociated

Jan  1 00:08:39 buildroot daemon.info hostapd: wlan0: STA f2:2e:1d:04:ae:c9 IEEE 802.11: disassociated

Jan  1 00:09:09 buildroot daemon.info hostapd: wlan0: STA f2:2e:1d:04:ae:c9 IEEE 802.11: disassociated

Jan  1 00:09:39 buildroot daemon.info hostapd: wlan0: STA f2:2e:1d:04:ae:c9 IEEE 802.11: disassociated

Jan  1 00:09:46 buildroot authpriv.info dropbear[289]: Child connection from 10.0.0.3:58858

Jan  1 00:10:00 buildroot authpriv.notice dropbear[289]: Password auth succeeded for 'root' from 10.0.0.3:58858

Jan  1 00:10:09 buildroot daemon.info hostapd: wlan0: STA f2:2e:1d:04:ae:c9 IEEE 802.11: disassociated

Jan  1 00:10:39 buildroot daemon.info hostapd: wlan0: STA f2:2e:1d:04:ae:c9 IEEE 802.11: disassociated

divyangmd, 2024-09-06:
let me know if its easier to paste the files into here.  I didn't do that this time since it takes up so much room on the thread.
Thanks again

sbala05, 2024-09-09:
My car (Audi A4) does not connect and I stumbled upon your issue and success and decided to try it. Previously the board was continuously rebooting, but now the attached logs repeat. Did you @divyangmd  or @nisargjhaveri see any clues? 
[a4log_9sep24.txt](https://github.com/user-attachments/files/16929026/a4log_9sep24.txt)

To be clear: If I only connect OTG cable, it reboots badly that I can't even grab logs. If I follow the sequence of connect both cables, pull out the OTG an put it back in the car, I get attached logs.

divyangmd, 2024-09-14:
Could someone with an understanding on how to decipher the logs take a look at my messages 2 and messages 3 logs above and  tell me what is happening during my process of connecting as I described in a post above.  Briefly I am able to get Android Auto wireless to work but only if I unplug and re-plug the OTG port in while supplying additional power through the power port of the rp Zero2w.  Once the Android Auto wireless activates on my truck I am able to unplug the power port and leave only the OTG plugged in to have it function. So the power port plug in is only necessary in the beginning.
Any help will be appreciated.  Thanks.

nisargjhaveri, 2024-09-19:
@divyangmd, the logs in messages3 suggests to some issue with USB, but I'm not able to say more in what exactly is going wrong and why it gets fixed by reconnecting the OTG port.

@sbala05, the logs suggests that the car never responds to the USB. Make sure you're using a good quality data cable, as that seems to be one of the most common issue and solution for similar problems.

#88 has more info on the issue where both cables are required. Have a look and see if anything there helps? We can also continue the conversation there.