nisargjhaveri/WirelessAndroidAutoDongle #173: Looking for android auto in Thar

srvsauravs, 2024-09-06:
Hi,

I installed it on RPi Zero W and used it on my Mahidnra Thar 2021. Firstly I connected to the AADongle Bluetooth and then to the AADongle Wifi. Somehow it starts connecting to the "Mahindra" Bluetooth i.e, the Cars bluetooth and Android Auto disconnects. It worked fine with my other Pixel 7 Pro even when connected to Mahindra Bluetooth. I keep forgetting the device and phone and delete the bluetooth Mahindra connection but it somehow pairs up again automatically.
Please suggest.

sasha7460, 2024-09-06:
@srvsauravs , does the wireless android auto works when using Pixel 7 Pro and connecting to both the RPi's and car's bluetooth?

srvsauravs, 2024-09-06:
@sasha7460  At first it did, I was amazed so wanted to try it on mine, but it didn't work for me and then neither on the Pixel 7 Pro. My other phone is One Plus 6 T

nisargjhaveri, 2024-09-06:
When Android Auto starts, it should automatically connect to the car bluetooth as well. And in most cases the dongle's bluetooth should disconnect.  That is normal. Can you please try to get the logs if the connection is not happening?

srvsauravs, 2024-09-09:
Hi @nisargjhaveri 

Can you please guide me as to how to extract the logs from the device?

srvsauravs, 2024-09-12:
> Jan  1 00:00:03 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:03 buildroot user.info console: Starting syslogd: OK
Jan  1 00:00:03 buildroot user.info console: Starting klogd: OK
Jan  1 00:00:03 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-08-03 03:46:38 UTC)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61 (runner@fv-az1490-423) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 Sat Aug  3 04:19:00 UTC 2024
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writeback
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:96:0D:09 vc_mem.mem_base=0x1ec0000
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory: 340196K/421888K available (7906K kernel code, 702K rwdata, 2184K rodata, 256K init, 498K bss, 16156K reserved, 65536K cma-reserved)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000006] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000056] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000157] bcm2835: system timer (irq = 27)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000946] Console: colour dummy device 80x30
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001446] printk: console [tty1] enabled
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001524] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060288] CPU: Testing write buffer coherency: ok
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060396] pid_max: default: 32768 minimum: 301
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060587] LSM: Security Framework initializing
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060797] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060868] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.062271] cgroup: Disabling memory control group subsystem
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064162] cblist_init_generic: Setting adjustable number of callback queues.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064246] cblist_init_generic: Setting shift to 0 and lim to 1.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064569] Setting up static identity map for 0x8220 - 0x8258
Jan  1 00:00:03 buildroot kern.info kernel: [    0.065821] devtmpfs: initialized
Jan  1 00:00:03 buildroot kern.info kernel: [    0.076662] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[... cut ...]
Jan  1 00:15:18 buildroot user.info aawgd[217]: Bluetooth device already connected, disconnecting
Jan  1 00:15:23 buildroot daemon.err bluetoothd[132]: src/profile.c:record_cb() Unable to get HSP HS SDP record: Host is down
Jan  1 00:15:29 buildroot kern.info kernel: [  929.018049] read descriptors
Jan  1 00:15:29 buildroot kern.info kernel: [  929.019487] read strings
Jan  1 00:15:38 buildroot user.info aawgd[217]: Failed to connect device at path: /org/bluez/hci0/dev_FC_AA_B6_EE_6E_7D
Jan  1 00:15:38 buildroot user.info aawgd[217]: Failed to connect to any known bluetooth device

srvsauravs, 2024-09-12:
Hi @nisargjhaveri 

This is my today's log. Somehow it just keeps disconnecting/switching between the two Bluetooth connections and disconnect.

> Jan  1 00:00:03 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:03 buildroot user.info console: Starting syslogd: OK
Jan  1 00:00:03 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-08-03 03:46:38 UTC)
Jan  1 00:00:03 buildroot user.info console: Starting klogd: OK
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61 (runner@fv-az1490-423) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 Sat Aug  3 04:19:00 UTC 2024
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writeback
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:96:0D:09 vc_mem.mem_base=0x1ec0000
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory: 340196K/421888K available (7906K kernel code, 702K rwdata, 2184K rodata, 256K init, 498K bss, 16156K reserved, 65536K cma-reserved)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000006] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000057] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000158] bcm2835: system timer (irq = 27)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000945] Console: colour dummy device 80x30
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001442] printk: console [tty1] enabled
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001520] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060288] CPU: Testing write buffer coherency: ok
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060395] pid_max: default: 32768 minimum: 301
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060585] LSM: Security Framework initializing
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060795] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060864] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.062263] cgroup: Disabling memory control group subsystem
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064141] cblist_init_generic: Setting adjustable number of callback queues.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064224] cblist_init_generic: Setting shift to 0 and lim to 1.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064544] Setting up static identity map for 0x8220 - 0x8258
Jan  1 00:00:03 buildroot kern.info kernel: [    0.065797] devtmpfs: initialized
Jan  1 00:00:03 buildroot kern.info kernel: [    0.076630] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[... cut ...]
Jan  1 00:01:42 buildroot daemon.info hostapd: wlan0: STA 88:53:2e:43:62:9b IEEE 802.11: disassociated
Jan  1 00:01:42 buildroot daemon.info hostapd: wlan0: STA 88:53:2e:43:62:9b IEEE 802.11: associated
Jan  1 00:01:42 buildroot daemon.info hostapd: wlan0: STA 88:53:2e:43:62:9b WPA: pairwise key handshake completed (RSN)
Jan  1 00:01:42 buildroot daemon.info dnsmasq-dhcp[180]: DHCPREQUEST(wlan0) 10.0.0.6 88:53:2e:43:62:9b 
Jan  1 00:01:42 buildroot daemon.info dnsmasq-dhcp[180]: DHCPACK(wlan0) 10.0.0.6 88:53:2e:43:62:9b Saurav-Laptop

nisargjhaveri, 2024-09-15:
Your second log suggests everything else works well, apart from the usb connection. Maybe double check the cable and the port you're using on the RPi? Make sure to use a good quality data cable (not power only cable) with the USB port in RPi 0 W (not the PWR port).

Other suspect is insufficient power somehow messing with bluetooth as bluetooth was not connecting well in your first log. But from the second log, I find it less likely.