nisargjhaveri/WirelessAndroidAutoDongle #174: 4.1 on Pi Zero 2 has no Bluetooth to connect to for initial set up

ShaggyStallion, 2024-09-07:
Title says it all... There's no Bluetooth to connect to for set up with 4.1 on a pi Zero 2

nisargjhaveri, 2024-09-07:
If no bluetooth or wifi is created, I'd suggest verifying the following:
- Make sure the correct image is used for the board you're using.
- Make sure that the image was correctly written to the SD card and the Pi boots. You should see flickers in the LED on your board.
- Make data cable is good quality and the power supplied by the car is enough.

ShaggyStallion, 2024-09-07:
Yes I have verified all of the steps mentioned. It was working with 4.0, although it developed a loop of crashing at times. I have not changed the cable and I verified that I am using the image for the zero 2 not the zero.

ShaggyStallion, 2024-09-07:
Solved!! I had to delete the saved Bluetooth connection for the device that was on my phone from when it was running the previous version of the software. Once I deleted it and tried booting it up, the Bluetooth connection showed up!

closed 2024-09-07 by ShaggyStallion