nisargjhaveri/WirelessAndroidAutoDongle #175: wifi not enabled

jatinderpalsingh, 2024-09-07:
I am using Raspberry PI Zero W. I flashed the image file on 16gb memory card. 
After flash, I connected to RPI0w to my car. Only Bluetooth is being detected in Phone as AADonglePlus. 
On connecting with bluetooth, it is connected for CALLS only. As far as I know it has to be CALLS and MEDIA connection only then Android Auto can work.
I manually checked WIFI networks. there is no wifi network showing with name AAWirelessDongle. It means wifi is not getting enabled.

I re-flashed the image. but the same issue, WIFI is not getting enabled. and phone is struck to "Looking for Android Auto".

I connected power supply as well usb otg to boost power to rpi0w, but it didn't make any difference.

nisargjhaveri, 2024-09-07:
"AADonglePlus" is not the name of the bluetooth device created by this project. It should be called "AndroidAuto-Dongle" or "AA Wireless Dongle" depending on the version you're using.

If no bluetooth or wifi is created, I'd suggest verifying the following:
- Make sure the correct image is used for the board you're using.
- Make sure that the image was correctly written to the SD card and the Pi boots. You should see flickers in the LED on your board.
- Make data cable is good quality and the power supplied by the car is enough.

jatinderpalsingh, 2024-09-10:
As Advised, I downloaded the pre-built image for RPI0W and flashed it using RPI Imager. Earlier I was using Balena Etcher to flash the image. I am using 16gb sd-card. I connected it to my car using OTG port. Green light flashed a bit. but then it was stable. Only bluetooth was discoverable as "AA Wireless Dongle". When connected with bluetooth, it is connected for Calls only, not media. I tried 2 different phones: Samsung s21 and Sony Xperia 10 vi. In both phones, it is connected for calls only. 
No wifi hotspot is shown. android auto is struck with notification "looking for android auto".

jatinderpalsingh, 2024-09-10:
so issue is not resolved yet. I tried giving separate power supply like I earlier did. but even that didn't work.
Is there any update required? I am using raspberry pi zero w v1.1