nisargjhaveri/WirelessAndroidAutoDongle #176: Can't get it to work at all on Raspberry Pi Zero 2 W (v0.4.1)

andrewjbennett, 2024-09-08:
I've been having problems trying to get any wireless AA dongle to work on a Toyota RAV4 2023 hybrid (wireless AA isn't available in the car by default in my country), so I figured I'd play around with this to see whether I can at least get any more debug info about what's going wrong.

I've flashed the image onto an sd card and I'm running it successfully in a Raspberry Pi Zero 2 W (successfully as in, it boots and I can SSH in etc), but I'm struggling to get anything working, even with the DHU, before I even get the car involved.

It looks like the phone can't maintain a connection with the Pi? I'm frequently getting messages like "Bluetooth device already connected, disconnecting" (not sure if that's a problem or expected?), as well as my phone repeatedly associating+disassociating from the Pi. Occasionally the phone will report an authentication problem with the wifi network, but it's clearly authenticating enough sometimes, since it gets a DHCP lease etc.

Snippet from /var/log/messages:

```
Jan  1 00:06:09 buildroot daemon.err bluetoothd[168]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
Jan  1 00:06:11 buildroot user.info aawgd[293]: HSP HS NewConnection
Jan  1 00:06:11 buildroot user.info aawgd[293]: Path: /org/bluez/hci0/dev_4C_2E_5E_D1_B6_3E, fd: 14
Jan  1 00:06:11 buildroot user.info aawgd[293]: Bluetooth connected to the device
Jan  1 00:06:11 buildroot user.info aawgd[293]: AA Wireless NewConnection
Jan  1 00:06:11 buildroot user.info aawgd[293]: Path: /org/bluez/hci0/dev_4C_2E_5E_D1_B6_3E, fd: 15
Jan  1 00:06:11 buildroot user.info aawgd[293]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:06:11 buildroot user.info aawgd[293]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:06:11 buildroot user.info aawgd[293]: Read WifiInfoRequest. length: 0, messageId: 2
Jan  1 00:06:11 buildroot user.info aawgd[293]: Sending WifiInfoResponse (ssid: AAWirelessDongle, bssid: 2c:cf:67:87:44:6a)
Jan  1 00:06:11 buildroot user.info aawgd[293]: Sent WifiInfoResponse, messageId: 3, wrote 70 bytes
Jan  1 00:06:11 buildroot user.info aawgd[293]: Read WifiStartResponse. length: 2, messageId: 7
Jan  1 00:06:12 buildroot daemon.info hostapd: wlan0: STA 66:ae:47:54:4c:7d IEEE 802.11: associated
Jan  1 00:06:16 buildroot daemon.info hostapd: wlan0: STA 66:ae:47:54:4c:7d IEEE 802.11: disassociated
Jan  1 00:06:16 buildroot daemon.info hostapd: wlan0: STA 66:ae:47:54:4c:7d IEEE 802.11: disassociated
Jan  1 00:06:16 buildroot daemon.info hostapd: wlan0: STA 66:ae:47:54:4c:7d IEEE 802.11: disassociated
Jan  1 00:06:16 buildroot daemon.info hostapd: wlan0: STA 66:ae:47:54:4c:7d IEEE 802.11: associated
Jan  1 00:06:20 buildroot daemon.info hostapd: wlan0: STA 66:ae:47:54:4c:7d IEEE 802.11: disassociated
Jan  1 00:06:21 buildroot user.info aawgd[293]: Read WifiConnectStatus. length: 11, messageId: 6
Jan  1 00:06:21 buildroot user.info aawgd[293]: Bluetooth launch sequence completed
Jan  1 00:06:31 buildroot user.info aawgd[293]: Found 2 bluetooth devices
Jan  1 00:06:31 buildroot user.info aawgd[293]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_4C_2E_5E_D1_B6_3E
Jan  1 00:06:31 buildroot user.info aawgd[293]: Bluetooth device already connected, disconnecting
Jan  1 00:06:31 buildroot user.info aawgd[293]: HSP HS RequestDisconnection
Jan  1 00:06:31 buildroot user.info aawgd[293]: Path: /org/bluez/hci0/dev_4C_2E_5E_D1_B6_3E
Jan  1 00:06:33 buildroot daemon.err bluetoothd[168]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
Jan  1 00:06:35 buildroot daemon.err bluetoothd[168]: src/profile.c:ext_connect() HSP HS failed connect to 4C:2E:5E:D1:B6:3E: Connection reset by peer (104)
Jan  1 00:06:35 buildroot user.info aawgd[293]: Failed to connect device at path: /org/bluez/hci0/dev_4C_2E_5E_D1_B6_3E
Jan  1 00:06:35 buildroot user.info aawgd[293]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_4C_6A_F6_4C_7F_C0
Jan  1 00:06:35 buildroot user.info aawgd[293]: Bluetooth device already connected, disconnecting
Jan  1 00:06:35 buildroot user.info aawgd[293]: AA Wireless NewConnection
Jan  1 00:06:35 buildroot user.info aawgd[293]: Path: /org/bluez/hci0/dev_4C_2E_5E_D1_B6_3E, fd: 16
Jan  1 00:06:35 buildroot user.info aawgd[293]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:06:35 buildroot user.info aawgd[293]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:06:35 buildroot user.info aawgd[293]: Read WifiInfoRequest. length: 0, messageId: 2
Jan  1 00:06:35 buildroot user.info aawgd[293]: Sending WifiInfoResponse (ssid: AAWirelessDongle, bssid: 2c:cf:67:87:44:6a)
Jan  1 00:06:35 buildroot user.info aawgd[293]: Sent WifiInfoResponse, messageId: 3, wrote 70 bytes
Jan  1 00:06:35 buildroot user.info aawgd[293]: Read WifiStartResponse. length: 2, messageId: 7
Jan  1 00:06:36 buildroot daemon.info hostapd: wlan0: STA 66:ae:47:54:4c:7d IEEE 802.11: associated
Jan  1 00:06:40 buildroot daemon.info hostapd: wlan0: STA 66:ae:47:54:4c:7d IEEE 802.11: disassociated
Jan  1 00:06:40 buildroot daemon.info hostapd: wlan0: STA 66:ae:47:54:4c:7d IEEE 802.11: disassociated
Jan  1 00:06:40 buildroot daemon.info hostapd: wlan0: STA 66:ae:47:54:4c:7d IEEE 802.11: disassociated
Jan  1 00:06:41 buildroot daemon.info hostapd: wlan0: STA 66:ae:47:54:4c:7d IEEE 802.11: associated
Jan  1 00:06:43 buildroot kern.info kernel: [  403.964099] read descriptors
Jan  1 00:06:43 buildroot kern.info kernel: [  403.965281] read strings
Jan  1 00:06:45 buildroot daemon.info hostapd: wlan0: STA 66:ae:47:54:4c:7d IEEE 802.11: disassociated
Jan  1 00:06:45 buildroot user.info aawgd[293]: Read WifiConnectStatus. length: 11, messageId: 6
Jan  1 00:06:45 buildroot user.info aawgd[293]: Bluetooth launch sequence completed
Jan  1 00:06:45 buildroot user.info aawgd[293]: HSP HS NewConnection
Jan  1 00:06:45 buildroot user.info aawgd[293]: Path: /org/bluez/hci0/dev_4C_6A_F6_4C_7F_C0, fd: 17
Jan  1 00:06:45 buildroot user.info aawgd[293]: AA Wireless NewConnection
Jan  1 00:06:45 buildroot user.info aawgd[293]: Path: /org/bluez/hci0/dev_4C_6A_F6_4C_7F_C0, fd: 19
Jan  1 00:06:45 buildroot user.info aawgd[293]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:06:45 buildroot user.info aawgd[293]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:06:46 buildroot user.info aawgd[293]: Read WifiInfoRequest. length: 0, messageId: 2
Jan  1 00:06:46 buildroot user.info aawgd[293]: Sending WifiInfoResponse (ssid: AAWirelessDongle, bssid: 2c:cf:67:87:44:6a)
Jan  1 00:06:46 buildroot user.info aawgd[293]: Sent WifiInfoResponse, messageId: 3, wrote 70 bytes
Jan  1 00:06:46 buildroot user.info aawgd[293]: Read WifiStartResponse. length: 2, messageId: 7
Jan  1 00:06:46 buildroot daemon.info hostapd: wlan0: STA 4c:6a:f6:cc:7f:c0 IEEE 802.11: disassociated
Jan  1 00:06:46 buildroot daemon.info hostapd: wlan0: STA 4c:6a:f6:cc:7f:c0 IEEE 802.11: disassociated
Jan  1 00:06:46 buildroot daemon.info hostapd: wlan0: STA 4c:6a:f6:cc:7f:c0 IEEE 802.11: disassociated
[... 71 lines cut ...]
Jan  1 00:07:50 buildroot user.info aawgd[293]: Path: /org/bluez/hci0/dev_4C_2E_5E_D1_B6_3E, fd: 21
Jan  1 00:07:50 buildroot user.info aawgd[293]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:07:50 buildroot user.info aawgd[293]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:07:50 buildroot user.info aawgd[293]: Read WifiInfoRequest. length: 0, messageId: 2
Jan  1 00:07:50 buildroot user.info aawgd[293]: Sending WifiInfoResponse (ssid: AAWirelessDongle, bssid: 2c:cf:67:87:44:6a)
Jan  1 00:07:50 buildroot user.info aawgd[293]: Sent WifiInfoResponse, messageId: 3, wrote 70 bytes
Jan  1 00:07:50 buildroot user.info aawgd[293]: Read WifiStartResponse. length: 2, messageId: 7
```

The DHU can't find the USB device (but also IIUC there would be different logs saying that it's enabled the USB, if it had made it that far?):

```
[I]: No device found ready yet, will retry shortly...
[I]: Found 11 USB devices.
[E]: Couldn't find/access compatible USB device.
[E]: Failed to start Google Automotive Link.
```

---

Any suggestions on what to try next? I've tried two different phones which have both had the same problem.

Is it a problem that I'm connecting to the wifi via my laptop first (to SSH in and view the logs) before I connect it to my phone?

Thanks!

Ashtray1807, 2024-09-08:
This happens for me also on a suzuki vitara 2017...

It connects to bluetooth but when it automatically tries to connect to wifi the phone screen flashes and it disconnects from Bluetooth

nisargjhaveri, 2024-09-08:
> The DHU can't find the USB device (but also IIUC there would be different logs saying that it's enabled the USB, if it had made it that far?):

Yes, it waits for the TCP connection from the phone before enabling usb, which never seems to happen in your case from the logs. Laptop being connected should not be a problem. Seems like after the bluetooth connection, wifi is connected, but then it disconnects. Make sure **not** to connect to the wifi manually on phone while it is connecting.

> It connects to bluetooth but when it automatically tries to connect to wifi the phone screen flashes and it disconnects from Bluetooth

@Ashtray1807  This looks correct and looks like wifi works but may be an issue with USB. Doesn't look like the same issue. Logs might help.

Ashtray1807, 2024-09-08:
Where can i get the logs from? Sorry, i'm new to this.

andrewjbennett, 2024-09-09:
@Ashtray1807 

> Where can i get the logs from? Sorry, i'm new to this.

You can get the logs by connecting via SSH to the Pi from your computer, either just reading them via SSH, or copying them to your computer via something like SCP.

----

The steps to do this:

_(disclaimer: I don't endorse any of the tutorial links below, I just found them on google and they seem reasonable)_

### To view the logs:

1. From your computer, connect to the `AAWirelessDongle` wifi network
2. Follow the instructions from somewhere like:
   * https://www.onlogic.com/blog/how-to-ssh-into-raspberry-pi/ (from the section "How to SSH into Raspberry Pi" onwards), or 
   *  https://voron.dozuki.com/Guide/Connecting+to+your+Raspberry+Pi+via+SSH/2#s4, from step 2 
3. The IP address of the Pi is `10.0.0.1`, the username is `root`, and the password is `password`
4. From there, you can type `cat /var/log/messages` to see all of the logs

### To download the logs:

1. Again, connect your computer to the `AAWirelessDongle` wifi network
2. Use something like WinSCP (on windows) or just `scp` on MacOS
   * some random tutorial links I found online: https://embeddedcomputing.com/technology/open-source/development-kits/easy-raspberry-pi-file-transfer-with-winscp for windows, https://spellfoundry.com/docs/copying-files-to-and-from-raspberry-pi-and-mac/#0-toc-title for mac
3. Again, the IP address of the Pi is `10.0.0.1`, the username is `root`, and the password is `password`
4. From there, you should be able to browse the files on the pi; the one you want to copy out is `/var/log/messages` (so there's a folder called `var`, and inside that folder there's one called `log`, and inside that there's a file called `messages`)

----

(I haven't checked whether there's already instructions for this somewhere on this github; if not feel free to put these, or an improved version of these, somewhere more visible)

andrewjbennett, 2024-09-09:
@nisargjhaveri 

> Yes, it waits for the TCP connection from the phone before enabling usb, which never seems to happen in your case from the logs. Laptop being connected should not be a problem. Seems like after the [...]

Ah cool, I was connecting to the wifi manually, so that probably caused problems.

Unfortunately it's still having problems with not being able to stay connected to the wifi network, even when I don't try to connect it myself.

More logs:

```
Jan  1 00:05:48 buildroot user.info aawgd[291]: AA Wireless Dongle
Jan  1 00:05:48 buildroot user.info aawgd[291]: Starting uevent monitoring
Jan  1 00:05:48 buildroot user.info aawgd[291]: Uevent monitoring started
Jan  1 00:05:48 buildroot user.info aawgd[291]: Initializing USB Manager
Jan  1 00:05:48 buildroot user.info aawgd[291]: USB Manager: Disabled all USB gadgets
Jan  1 00:05:48 buildroot user.info aawgd[291]: USB Manager: Found UDC 3f980000.usb
Jan  1 00:05:48 buildroot user.info aawgd[291]: Using bluetooth adapter at path: /org/bluez/hci0
Jan  1 00:05:48 buildroot user.info aawgd[291]: Bluetooth AA Wireless profile active
Jan  1 00:05:48 buildroot user.info aawgd[291]: HSP Handset profile active
Jan  1 00:05:48 buildroot user.info aawgd[291]: Connection Strategy: 1
Jan  1 00:05:48 buildroot user.info aawgd[291]: Starting tcp server
Jan  1 00:05:48 buildroot user.info aawgd[291]: Tcp server listening on 5288
Jan  1 00:05:48 buildroot user.info aawgd[291]: Bluetooth adapter was powered on
Jan  1 00:05:48 buildroot user.info aawgd[291]: Bluetooth adapter is now discoverable and pairable
Jan  1 00:05:48 buildroot user.info aawgd[291]: Found 2 bluetooth devices
Jan  1 00:05:48 buildroot user.info aawgd[291]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_4C_2E_5E_D1_B6_3E
Jan  1 00:05:48 buildroot user.info aawgd[291]: Bluetooth device already connected, disconnecting
Jan  1 00:05:49 buildroot user.info aawgd[291]: HSP HS NewConnection
Jan  1 00:05:49 buildroot user.info aawgd[291]: Path: /org/bluez/hci0/dev_4C_6A_F6_4C_7F_C0, fd: 11
Jan  1 00:05:50 buildroot user.info aawgd[291]: AA Wireless NewConnection
Jan  1 00:05:50 buildroot user.info aawgd[291]: Path: /org/bluez/hci0/dev_4C_6A_F6_4C_7F_C0, fd: 12
Jan  1 00:05:50 buildroot user.info aawgd[291]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:05:50 buildroot user.info aawgd[291]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:05:50 buildroot user.info aawgd[291]: Read WifiInfoRequest. length: 0, messageId: 2
Jan  1 00:05:50 buildroot user.info aawgd[291]: Sending WifiInfoResponse (ssid: AAWirelessDongle, bssid: 2c:cf:67:87:44:6a)
Jan  1 00:05:50 buildroot user.info aawgd[291]: Sent WifiInfoResponse, messageId: 3, wrote 70 bytes
Jan  1 00:05:50 buildroot user.info aawgd[291]: Read WifiStartResponse. length: 2, messageId: 7
Jan  1 00:05:54 buildroot daemon.err bluetoothd[165]: src/profile.c:record_cb() Unable to get HSP HS SDP record: Host is down
Jan  1 00:06:01 buildroot daemon.info hostapd: wlan0: STA 8e:09:aa:23:1d:89 IEEE 802.11: associated
Jan  1 00:06:03 buildroot daemon.info hostapd: wlan0: STA 8e:09:aa:23:1d:89 IEEE 802.11: disassociated
Jan  1 00:06:03 buildroot daemon.info hostapd: wlan0: STA 8e:09:aa:23:1d:89 IEEE 802.11: disassociated
Jan  1 00:06:03 buildroot daemon.info hostapd: wlan0: STA 8e:09:aa:23:1d:89 IEEE 802.11: associated
Jan  1 00:06:03 buildroot kern.info kernel: [  363.924078] read descriptors
Jan  1 00:06:03 buildroot kern.info kernel: [  363.925268] read strings
Jan  1 00:06:04 buildroot daemon.info hostapd: wlan0: STA 44:af:28:3d:66:cf IEEE 802.11: disassociated
Jan  1 00:06:04 buildroot daemon.info hostapd: wlan0: STA 44:af:28:3d:66:cf IEEE 802.11: disassociated
Jan  1 00:06:05 buildroot daemon.info hostapd: wlan0: STA 44:af:28:3d:66:cf IEEE 802.11: associated
Jan  1 00:06:05 buildroot daemon.info hostapd: wlan0: STA 44:af:28:3d:66:cf IEEE 802.11: disassociated
Jan  1 00:06:05 buildroot daemon.info hostapd: wlan0: STA 44:af:28:3d:66:cf IEEE 802.11: disassociated
Jan  1 00:06:05 buildroot daemon.info hostapd: wlan0: STA 44:af:28:3d:66:cf IEEE 802.11: disassociated
Jan  1 00:06:06 buildroot daemon.info hostapd: wlan0: STA 44:af:28:3d:66:cf IEEE 802.11: associated
Jan  1 00:06:06 buildroot daemon.info hostapd: wlan0: STA 44:af:28:3d:66:cf WPA: pairwise key handshake completed (RSN)
Jan  1 00:06:06 buildroot daemon.info dnsmasq-dhcp[224]: DHCPREQUEST(wlan0) 10.0.0.19 44:af:28:3d:66:cf 
Jan  1 00:06:06 buildroot daemon.info dnsmasq-dhcp[224]: DHCPACK(wlan0) 10.0.0.19 44:af:28:3d:66:cf briar
Jan  1 00:06:07 buildroot daemon.info hostapd: wlan0: STA 8e:09:aa:23:1d:89 IEEE 802.11: disassociated
Jan  1 00:06:07 buildroot daemon.info hostapd: wlan0: STA 8e:09:aa:23:1d:89 IEEE 802.11: disassociated
Jan  1 00:06:08 buildroot user.info aawgd[291]: Failed to connect device at path: /org/bluez/hci0/dev_4C_2E_5E_D1_B6_3E
Jan  1 00:06:08 buildroot user.info aawgd[291]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_4C_6A_F6_4C_7F_C0
Jan  1 00:06:15 buildroot daemon.err bluetoothd[165]: src/profile.c:new_conn_reply() AA Wireless replied with an error: org.freedesktop.DBus.Error.NoReply, Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus secur
Jan  1 00:06:17 buildroot daemon.info hostapd: wlan0: STA 8e:09:aa:23:1d:89 IEEE 802.11: associated
Jan  1 00:06:21 buildroot daemon.info hostapd: wlan0: STA 8e:09:aa:23:1d:89 IEEE 802.11: disassociated
[... cut ...]
It seems a bit sus that I can see these logs about how my laptop is allegedly disassociating from the wifi, while it's still connected enough to the wifi to be able to keep tailing the logs in real time... but definitely on my phone it seems to be having a hard time staying connected. 

I've tried two different phones for this already, I'll see if I can find any other phones to try it on. 

I'm suspicious that it might be unhappy with the AA wifi network not having internet available, but I've explicitly told my phone to allow it to use that network even though it doesn't have internet, which I would have thought would solve that problem?

andrewjbennett, 2024-09-09:
I removed all of the bluetooth devices from the Pi, and I'm now trying on a third phone, which is maybe making some progress? But still not working....

It's clearly in some sort of loop, looking at just the aawgd logs it shows this happening in a loop:

```
aawgd[290]: Found 1 bluetooth devices
aawgd[290]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_D4_3A_2C_9C_15_1B
aawgd[290]: Bluetooth device already connected, disconnecting
aawgd[290]: HSP HS NewConnection
aawgd[290]: Path: /org/bluez/hci0/dev_D4_3A_2C_9C_15_1B, fd: 29
aawgd[290]: Bluetooth connected to the device
aawgd[290]: AA Wireless NewConnection
aawgd[290]: Path: /org/bluez/hci0/dev_D4_3A_2C_9C_15_1B, fd: 30
aawgd[290]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
aawgd[290]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
aawgd[290]: Read WifiInfoRequest. length: 0, messageId: 2
aawgd[290]: Sending WifiInfoResponse (ssid: AAWirelessDongle, bssid: 2c:cf:67:87:44:6a)
aawgd[290]: Sent WifiInfoResponse, messageId: 3, wrote 70 bytes
aawgd[290]: Read WifiStartResponse. length: 2, messageId: 7
aawgd[290]: Read WifiConnectStatus. length: 11, messageId: 6
aawgd[290]: Bluetooth launch sequence completed
```

When it gets to the "Bluetooth launch sequence completed", the phone has an AA popup like "to continue, select Android Auto on your car screen", then after a few seconds it stops showing on the phone, and aawgd goes back to the start of the loop: `Found 1 bluetooth devices`

some /var/log/messages:

```
Jan  1 00:05:36 buildroot user.info aawgd[290]: Found 1 bluetooth devices
Jan  1 00:05:36 buildroot user.info aawgd[290]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_D4_3A_2C_9C_15_1B
Jan  1 00:05:36 buildroot user.info aawgd[290]: Bluetooth device already connected, disconnecting
Jan  1 00:05:36 buildroot user.info aawgd[290]: HSP HS RequestDisconnection
Jan  1 00:05:36 buildroot user.info aawgd[290]: Path: /org/bluez/hci0/dev_D4_3A_2C_9C_15_1B
Jan  1 00:05:39 buildroot daemon.err bluetoothd[167]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
Jan  1 00:05:40 buildroot daemon.err bluetoothd[167]: src/profile.c:record_cb() Unable to get HSP HS SDP record: Operation already in progress
Jan  1 00:05:40 buildroot user.info aawgd[290]: Failed to connect device at path: /org/bluez/hci0/dev_D4_3A_2C_9C_15_1B
Jan  1 00:05:40 buildroot user.info aawgd[290]: Failed to connect to any known bluetooth device
Jan  1 00:05:40 buildroot user.info aawgd[290]: AA Wireless NewConnection
Jan  1 00:05:40 buildroot user.info aawgd[290]: Path: /org/bluez/hci0/dev_D4_3A_2C_9C_15_1B, fd: 22
Jan  1 00:05:40 buildroot user.info aawgd[290]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:05:40 buildroot user.info aawgd[290]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:05:40 buildroot user.info aawgd[290]: Read WifiInfoRequest. length: 0, messageId: 2
Jan  1 00:05:40 buildroot user.info aawgd[290]: Sending WifiInfoResponse (ssid: AAWirelessDongle, bssid: 2c:cf:67:87:44:6a)
Jan  1 00:05:40 buildroot user.info aawgd[290]: Sent WifiInfoResponse, messageId: 3, wrote 70 bytes
Jan  1 00:05:40 buildroot user.info aawgd[290]: Read WifiStartResponse. length: 2, messageId: 7
Jan  1 00:05:43 buildroot daemon.info hostapd: wlan0: STA 46:e4:ff:eb:2b:ff IEEE 802.11: associated
Jan  1 00:05:47 buildroot daemon.info hostapd: wlan0: STA 46:e4:ff:eb:2b:ff IEEE 802.11: disassociated
Jan  1 00:05:50 buildroot daemon.info hostapd: wlan0: STA 46:e4:ff:eb:2b:ff IEEE 802.11: associated
Jan  1 00:05:54 buildroot daemon.info hostapd: wlan0: STA 46:e4:ff:eb:2b:ff IEEE 802.11: disassociated
Jan  1 00:05:55 buildroot user.info aawgd[290]: Read WifiConnectStatus. length: 11, messageId: 6
Jan  1 00:05:55 buildroot user.info aawgd[290]: Bluetooth launch sequence completed
Jan  1 00:06:00 buildroot user.info aawgd[290]: Found 1 bluetooth devices
Jan  1 00:06:00 buildroot user.info aawgd[290]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_D4_3A_2C_9C_15_1B
Jan  1 00:06:00 buildroot user.info aawgd[290]: Bluetooth device already connected, disconnecting
Jan  1 00:06:02 buildroot daemon.err bluetoothd[167]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
Jan  1 00:06:03 buildroot user.info aawgd[290]: HSP HS NewConnection
Jan  1 00:06:03 buildroot user.info aawgd[290]: Path: /org/bluez/hci0/dev_D4_3A_2C_9C_15_1B, fd: 23
Jan  1 00:06:03 buildroot user.info aawgd[290]: Bluetooth connected to the device
Jan  1 00:06:03 buildroot user.info aawgd[290]: AA Wireless NewConnection
Jan  1 00:06:03 buildroot user.info aawgd[290]: Path: /org/bluez/hci0/dev_D4_3A_2C_9C_15_1B, fd: 24
Jan  1 00:06:03 buildroot user.info aawgd[290]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:06:03 buildroot user.info aawgd[290]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:06:03 buildroot user.info aawgd[290]: Read WifiInfoRequest. length: 0, messageId: 2
Jan  1 00:06:03 buildroot user.info aawgd[290]: Sending WifiInfoResponse (ssid: AAWirelessDongle, bssid: 2c:cf:67:87:44:6a)
Jan  1 00:06:03 buildroot user.info aawgd[290]: Sent WifiInfoResponse, messageId: 3, wrote 70 bytes
Jan  1 00:06:03 buildroot user.info aawgd[290]: Read WifiStartResponse. length: 2, messageId: 7
[... cut ...]
Jan  1 00:08:06 buildroot daemon.info hostapd: wlan0: STA 46:e4:ff:eb:2b:ff IEEE 802.11: associated
Jan  1 00:08:10 buildroot daemon.info hostapd: wlan0: STA 46:e4:ff:eb:2b:ff IEEE 802.11: disassociated
Jan  1 00:08:11 buildroot daemon.info hostapd: wlan0: STA 46:e4:ff:eb:2b:ff IEEE 802.11: associated
```

Note that once the thing pops up on my phone, I try connecting DHU on my laptop, which just says `[I]: No device found ready yet, will retry shortly...` (but I also think that there'd be a different message in the logs if it had gotten the USB part ready?)

Ashtray1807, 2024-09-09:
@andrewjbennett when trying to connect to AAWirelessDongle it asks me for a password
[image]

andrewjbennett, 2024-09-09:
@Ashtray1807 

> when trying to connect to AAWirelessDongle it asks me for a password 

Have a look in the readme file (the main github page): https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/?tab=readme-ov-file#getting-logs

> Connect to the device using wifi (SSID:AAWirelessDongle, Password: ConnectAAWirelessDongle, see [hostapd.conf](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/main/aa_wireless_dongle/board/common/rootfs_overlay/etc/hostapd.conf)).

BTW, since it sounds like your problem isn't exactly the same as mine, you should probably create a new issue for it, rather than reusing this one.

reach2rv, 2024-09-09:
I am having the same issue. phone shows looking for Android auto and wifi keeps disconnecting when in connects screen flicker and it again shows looking for android Auto

andrewjbennett, 2024-09-10:
Update: It works!!! 

I built it from HEAD rather than using the v0.4.1 image, and needed to have both the data micro usb plugged into the car, as well as extra power plugged into the power micro usb port. 

Without the extra power it partially worked, but gave an error when trying to actually connect to the car. With the extra power, the problems went away.

I'll do a diff between my build and the v0.4.1 release to see what might be the cause.

andrewjbennett, 2024-09-10:
Update 2: I flashed the v0.4.1 image onto the SD card (rather than the one I built myself), and.... now it works too? I'm quite confused.

---

Here are the exact steps that I followed:

Initial laptop setup:
1. Download `sdkmanager` from the android SDK `cmdline-tools` thing
2. Install the android auto DHU: `./sdkmanager  --sdk_root=~/src/android/commandlinetools/cmdline-tools 'extras;google;auto'` 

Initial phone setup:
1. Remove all existing/saved connections/devices related to this (so forget the "AAWirelessDongle" wifi network, and delete the "AA Wireless Dongle" bluetooth device)
2. Plug my phone into my laptop
3. Run DHU via `/src/android/commandlinetools/cmdline-tools/extras/google/auto/desktop-head-unit  -u` (which was installed in laptop step 2 before)
4. Observe that it detects my phone, and that after ~20 seconds or so of initializing, it pops up with the android auto UI, showing me google maps and my phone's GPS location
5. Close DHU, and unplug my phone

Initial Pi setup:
1. Obtain two micro USB cables (high quality ones, which can do data)
2. Flash the (previously-downloaded) image onto an sd card (the same SD card as I started with):
  * ` $ sudo dd if=/home/andrew/src/dad/raspberrypizero2w-sdcard.img of=/dev/sda status=progress`
3. Put the SD card into the pi (specifically, a ["Raspberry Pi Zero 2 W (SC1176)"](https://www.digikey.com/en/products/detail/raspberry-pi/SC1176/15298147)

The steps from there:
1. Plug a micro USB cable from the Pi's data port to my laptop (which powers on the device)
2. Plug a micro USB cable from the Pi's power port to a USB power supply (which adds extra power)
3. From my laptop, switch from my home wifi to the `AAWirelessDongle` wifi network, and ssh into it (`ssh root@10.0.0.1`), then watch the logs (`tail -f /var/log/messages`)
4. On my phone (which has no saved connections) connect to the `AA Wireless Dongle` Bluetooth device that shows up
5. Don't touch anything else on my phone. Don't manually join any wifi networks, don't change any settings related to anything. Just leave it alone.
6. In the logs that I'm watching on my laptop (via SSH), observe it doing the bluetooth connection to the phone, then setting up the wifi, then starting the USB
7. Run DHU again, and observe it finding the Pi (`[I]: Device 'My Own AA Wireless Dongle 0123456' (vid=18d1, pid=4ee1) supports AOAPv2, starting accessory mode...`) and initializing etc
8. Don't touch anything else
9. Observe that my phone now shows up, the same as it did before when plugging it into my laptop via usb cable

---

My best guess for what happened here (in terms of why it didn't work the first time, but it did work when I flashed the exact same image onto the exact same SD card again) is that I inadvertently messed up some internal state or config etc, with the various things I tried to do to get it to work?

nisargjhaveri, 2024-09-12:
Thanks @andrewjbennett for experimenting and reporting in detail. Glad that you were able to make it work.

I don't know why it wouldn't work with your first flash.

For having to have two cables connected, can you also experiment with a couple of suggestions in https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/88#issuecomment-2332544722? If we can figure out a solution for this as well, it would help a lot.