nisargjhaveri/WirelessAndroidAutoDongle #178: Wifi + Bluetooth not connecting on Hyundai vehicle

anuragdev101, 2024-09-09:
I have an exter 2024 Model, the board I'm using is Raspberry Pi Zero Wireless. I'm having following issues:

1. Wifi not connecting 
2. Bluetooth not connecting 

Observations:
1. When connecting usb cable to usb port of PI, Wifi appears but doesn't connect on password.
2. Same behaviour with Bluetooth, it doesn't even shows in list. Pi led lights goes constantly on and off.
3. When connecting usb cable to power port of PI, Wifi appears and gets connected with same password.
4. Same behaviour with Bluetooth, it gets connected and then gets disconnected instantly after 1 or 2 second.

Could you please help in what could be the issue.

anuragdev101, 2024-09-10:
Problem is, my car only has 1 USB port

sasha7460, 2024-09-10:
@anuragdev101 , you could use a power bank, but @selfdestructer, as per the instructions, you do not have to power both the phone and the pi simultaneously, the android auto in the phone should be started before connecting the pi and the phone could be disconnected after this.

nisargjhaveri, 2024-09-19:
#88 has more info on the insufficient power issue. We don't have a solution yet, but it contains some suggestion you can try to lower the power consumption and see if it makes it work on your car.

AkshatAgrawal05, 2025-04-18:
Hi @anuragdev101 , is your issue resolved? I am also facing the same issue with Exter.