nisargjhaveri/WirelessAndroidAutoDongle #179: Raspberry Pi Zero 2 W image: switch to 64-bit

manio, 2024-09-09:
Hi,
Currently the **Pi Zero 2 W** build is a 32 bit, while the processor is 64-bit capable. I think it might work with better efficiency if the prepared image would also cover this. Can it be adjusted?

Currently the ELF binaries are:
`ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3`
While it could be:
`ELF 64-bit LSB pie executable, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1`

nisargjhaveri, 2024-09-12:
It probably can be changed, I believe #129 contains an attempt. But apparently it has some issues.

Also, I'm not sure if the Cpu efficiency is an issue in the first place for most cases.

manio, 2024-09-12:
Ah, thanks. If there's a PR for this I am closing this one.

closed 2024-09-12 by manio