nisargjhaveri/WirelessAndroidAutoDongle #182: Keeps on disconnecting the RPi z 2 w and Spotify Streaming Song Distortion - Altroz

SurajYOLO, 2024-09-11:
I was successfully able to launch the android auto on my Altroz, not very fluidic UI, Spotify songs or amazon music songs usually faces breaks in between and sometimes the songs are skipping 2 to 3 secs in between. Facing frequent disconnections and reconnections.

sasha7460, 2024-09-12:
hi @SurajYOLO, mine is an altroz, but i was never able to get the android auto started in the Head Unit after connecting pi via usb. 
could you elaborate the steps you followed to get the connection?  it could help me in case i missed something

SurajYOLO, 2024-09-12:
Hi @sasha7460, first thing is do you see any green light blinking on the Pi? if yes, do you see any adnroid auto searching loop on your phone? you should have a best in condition SD card flashed with the image file, a high speed and high quality usb to micro usb cable and the micro usb should be inserted to the USB point (Not on the PWR Point) on the Rasp Pi.

SurajYOLO, 2024-09-12:
My issue is almost solved by Duracell usb to micro usb cable. But what's the RF range of the Pi z 2 w? Do i need to keep my phone very next to the Pi?

SurajYOLO, 2024-09-13:
Hi @sasha7460  did it work for you?

nisargjhaveri, 2024-09-13:
> But what's the RF range of the Pi z 2 w? Do i need to keep my phone very next to the Pi?

The BT and Wifi range is generally fairly good. There should be no problem anywhere inside the car at all, it should be generally much larger than that.

SurajYOLO, 2024-09-13:
@nisargjhaveri  still i feel some jitter while playing music, would it be RPi issue?

sasha7460, 2024-09-14:
> Hi @sasha7460 did it work for you?

@SurajYOLO , not in my altroz.
the pi works in a different car.
 i was able to run the android auto in an innova hycross, but not on my tata altroz.
 this means both the cable and the sd card are good and the flashing is done correctly.
that's why i wanted to check if you followed any different step in getting it running on your altroz.
as far as i see only the board is different. i am using pi 0w and you are using pi 2 w.
@nisargjhaveri , would you know where to look for to identify what could be the problem?

codermvn, 2024-09-14:
I am using Tata Nexon EV Max 2.0. Used this setup but an facing disconnection whenever there is a sharp bump on the road. The Mobile keeping connecting to 'Android Auto' and on car I see 'Device Not Found' 3-4 times after which the connection never gets re established. How shall I diagnose further? 
Wired AA works well but my phone has type C cable but Raspberry Pi 0 2w has micro USB, so the connection medium isnt same

codermvn, 2024-09-14:
[image]
[image]

samirkape, 2024-09-15:
Exact same observation on elite i20. I thought it was car plug issue

SurajYOLO, 2024-09-15:
@samirkape how about normal jitters while playing music from Spotify?

samirkape, 2024-09-16:
I use Youtube Music, and obsered jitters happen when I am in a crowded place or near Wifi router (maybe because of 2.4Ghz interference?). It is not that frequent, though.

nisargjhaveri, 2024-09-19:
> The moment i go on a speed breaker then it disconnects

This certainly does seem like a cable or connection issue. @codermvn, the issue you mentioned also sounds like either bad cable or bad connection with the cable on either side. You can maybe try with a good quality data cable with firm connection on both ends.

SurajYOLO, 2024-09-19:
Hi @nisargjhaveri , can you tell me how to get the logs from the pi? I am not getting how to check the logs of issues. May be that will help you to suggest what is actually happening

SurajYOLO, 2024-09-30:
> This certainly does seem like a cable or connection issue. @codermvn, the issue you mentioned also sounds like either bad cable or bad connection with the cable on either side. You can maybe try [...]

@nisargjhaveri please guide me how to get the logs? That might help you suggest the reason for disconnections.

SurajYOLO, 2025-02-04:
Hi @nisargjhaveri , finally i got the logs. please see and suggest what is the issue of frequent disconnections. not sure if it has captured my car logs. 

Last login: Tue Feb  4 22:42:09 on ttys000
surajmoningi@Surajs-Air ~ % ssh root@10.0.0.1
The authenticity of host '10.0.0.1 (10.0.0.1)' can't be established.
ED25519 key fingerprint is SHA256:w9kQQF3zxLtp8iuWg6biPrHTnuLghy8iIBlniuMWChg.
This key is not known by any other names.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '10.0.0.1' (ED25519) to the list of known hosts.
root@10.0.0.1's password: 
Permission denied, please try again.
root@10.0.0.1's password: 
# password
-sh: password: not found
# 10.0.0.1
-sh: 10.0.0.1: not found
# /var/log/messages
-sh: /var/log/messages: Permission denied
# raspberrypi0w1_defconfig
-sh: raspberrypi0w1_defconfig: not found
# ssh raspberrypi0w1_defconfig

ssh: Connection to root@raspberrypi0w1_defconfig:22 exited: Connect failed: Error resolving 'raspberrypi0w1_defconfig' port '22'. Name or service not known
# cat /var/log/messages
Jan  1 00:00:05 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:05 buildroot user.info console: Starting syslogd: OK
Jan  1 00:00:05 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-08-03 03:46:23 UTC)
Jan  1 00:00:05 buildroot user.info console: Starting klogd: OK
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:05 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61-v7 (runner@fv-az1424-844) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP Sat Aug  3 04:19:00 UTC 2024
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero 2 W Rev 1.0
Jan  1 00:00:05 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000]   Normal   empty
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] percpu: Embedded 12 pages/cpu s18004 r8192 d22956 u49152
Jan  1 00:00:05 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s18004 r8192 d22956 u49152 alloc=12*4096
Jan  1 00:00:05 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:05 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:CA:24:62 vc_mem.mem_base=0x1ec0000
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] Memory: 337224K/421888K available (9216K kernel code, 718K rwdata, 2316K rodata, 1024K init, 520K bss, 19128K reserved, 65536K cma-reserved)
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
Jan  1 00:00:05 buildroot kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[... cut ...]
Jan  1 00:21:35 buildroot user.info aawgd[264]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_40_FA_FE_C9_EE_C1
Jan  1 00:21:35 buildroot user.info aawgd[264]: Bluetooth device already connected, disconnecting
Jan  1 00:21:40 buildroot daemon.err bluetoothd[166]: src/profile.c:record_cb() Unable to get HSP HS SDP record: Host is down
Jan  1 00:21:55 buildroot user.info aawgd[264]: Failed to connect device at path: /org/bluez/hci0/dev_40_FA_FE_C9_EE_C1
Jan  1 00:21:55 buildroot user.info aawgd[264]: Failed to connect to any known bluetooth device
#

snaveenkumareee, 2025-03-06:
I don't think it is because of speed breaker.. the connection drops even at the signals. this is because when RPi comes close to any another bluetooth. I requested to stop scanning for new device when device is already connected. 

if this is addressed, connection drops might get reduced. 
need feedback from dev if the above can be achieved.

manio, 2025-03-06:
@snaveenkumareee the bluetooth on the Pi is off during normal AndroidAuto working. It is needed only for initiating connection for a short time period.

manio, 2025-03-06:
> bluetooth may get disconnected from mobile.. but will it keep scanning for new mobile connection ?

the bluetooth is probably connected to the car as a headset. Regading scanning - I am not a bluetooth expert but I think it is nothing extraordinary - like in normal android phone state. Why should it scanning? It is connected to car.

> RPi connection drops when I stop by at the traffic signals.. what could be the problem? am I missing any?

Same place every time or a random location?

SurajYOLO, 2025-03-06:
I've seen disconnections when I'm taking a immediate acceleration and due
to that jerk, i feel there is a disconnection.

manio, 2025-03-06:
> when I'm taking a immediate acceleration

OK ... but to be sure - it has nothing common with any physical connection, moving smartphone closer to something, etc?

JBiegas611, 2026-09-02:
Thanks forr your work here!
On my wifes phone android auto seems to disconnect when the kia unit tries to reconnect her phone via blutooth again after starting android auto through the pi.  Any quick thoughts if this is why and how to stop it from happening?  To my understand the bt connection isnt even necessary after inital connect, so darn it for thebtime spent trying to figure it out.